RfidResearchGroup/proxmark3 #505: Script mfu_magic gives #db# unknown command:: 0xffff for ntag213

pp614, 2019-12-31:
**Describe the bug**
Running mfu_magic script for NXP 213 command on osx or Windows gives me error unknown command:: 0xffff.

```
[usb] pm3 --> script run mfu_magic -u 04112233445566
[+] Executing Lua script: /Volumes/B/proxmark3_ice/client/luascripts/mfu_magic.lua, args '-u 04112233445566'

----------------------------------------
----------------------------------------

Writing new UID 	04112233445566
#db# unknown command:: 0xffff
ERROR: 	Error, waiting for response timed out :: No response from the device
#db# unknown command:: 0xffff
ERROR: 	Error, waiting for response timed out :: No response from the device
#db# unknown command:: 0xffff
ERROR: 	Error, waiting for response timed out :: No response from the device

[+] Finished mfu_magic
```

**To Reproduce**
Steps to reproduce the behavior:
script run mfu_magic -u 04112233445566

**Expected behavior**
Change UID of ntag213 

**Desktop (please complete the following information):**
 - OS: macOS High Sierra
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version

```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman
  compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.11.45.5) OS:OSX ARCH:x86_64

 [ PROXMARK3 ]

 [ ARM ]
  bootrom: RRG/Iceman/master/7dc65bf0 2019-12-31 16:15:27
       os: RRG/Iceman/master/7dc65bf0 2019-12-31 16:15:46
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 281528 bytes (54%) Free: 242760 bytes (46%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

```
 
- hw status
```

[usb] pm3 --> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................0
#db#   traceLen ...............302
#db# Currently loaded FPGA image
#db#   mode.................... HF image built for 2s30vq100 on 2018-09-03 at 21:40:23
#db# Flash memory
#db#   Baudrate................24 MHz
#db#   Init....................FAILED
#db# Smart card module (ISO 7816)
#db#   version.................FAILED
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125.00 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db#   [s] samples to skip.....0
#db# LF T55XX config
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
#db#            mode            |start|write|write|write| read|write|write
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
#db#     long leading reference |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
#db#               leading zero |  29 |  17 |  15 |  40 |  15 | N/A | N/A |
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
#db#
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............500ms
#db#   Bytes transferred.......353280
#db#   Transfer Speed PM3 -> Client = 706560 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............39
#db#   ToSendBit...............8
#db#   ToSend BUFFERSIZE.......2308
#db#   Slow clock..............30651 Hz
#db# Installed StandAlone Mode
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[usb] pm3 -->

```

 - data tune
```

[usb] pm3 -->  data tune

[=] Measuring antenna characteristics, please wait...

[=] You can cancel this operation by pressing the pm3 button
..

[+] LF antenna: 26.91 V - 125.00 kHz
[+] LF antenna: 34.13 V - 134.83 kHz
[+] LF optimal: 34.70 V - 131.87 kHz
[+] LF antenna is OK

[+] HF antenna: 26.51 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 88 is 134.83 kHz, 95 is 125.00 kHz.

No GUI in this build!`

`[usb] pm3 --> hf 14a info

[+]  UID : 04 22 1D CA A3 65 81
[+] ATQA : 00 44
[+]  SAK : 00 [2]
TYPE : NTAG 213 144bytes (NT2H1311G0DU)
[+] MANUFACTURER : NXP Semiconductors Germany
[=] proprietary non iso14443-4 card found, RATS not supported

```

```

[usb] pm3 --> script run mfu_magic
[+] Executing Lua script: /Volumes/B/proxmark3_ice/client/luascripts/mfu_magic.lua, args ''

----------------------------------------
----------------------------------------

Copyright (c) 2017 IceSQL AB. All rights reserved.
Christian Herrmann
v1.1.1
This script enables easy programming of a MAGIC NTAG 21* card
Example usage
	-- wipe tag
	script run mfu_magic -w

	-- wipe a locked down tag by giving the password
	script run mfu_magic -k ffffffff -w

	--read magic tag configuration
	script run mfu_magic -c

	-- set uid
	script run mfu_magic -u 04112233445566

	-- set pwd / pack
	script run mfu_magic -p 11223344 -a 8080

	-- set version to NTAG213
	script run mfu_magic -v 0004040201000f03

	-- set signature
	script run mfu_magic -s 1122334455667788990011223344556677889900112233445566778899001122

Usage:
script run mfu_magic -h -k <passwd> -c -w -u <uid> -t <type> -p <passwd> -a <pack> -s <signature> -o <otp> -v <version>

Arguments:
	-h		this help
	-c		read magic configuration
	-u		UID (14 hexsymbols), set UID on tag
	-t		tag type to impersonate
				 1 = UL_EV1 48k
				 2 = UL_EV1 128k
				 3 = NTAG 210
				 4 = NTAG 212
				 5 = NTAG 213 (true)
				 6 = NTAG 215 (true)
				 7 = NTAG 216 (true)
				 8 = NTAG I2C 1K
				 9 = NTAG I2C 2K
				10 = NTAG I2C 1K PLUS
				11 = NTAG I2C 2K PLUS
				12 = NTAG 213F (true)
				13 = NTAG 216F (true)
	-p		password (8 hexsymbols),  set password on tag.
	-a		pack ( 4 hexsymbols), set pack on tag.
	-s		signature data (64 hexsymbols), set signature data on tag.
	-o		OTP data (8 hexsymbols), set one-time-pad data on tag.
	-v		version data (16 hexsymbols), set version data on tag.
	-w		wipe tag. You can specify password if the tag has been locked down. Fills tag with zeros and put default values for NTAG213 (like -t 5)
	-k		pwd to use with the wipe option

[+] Finished mfu_magic`

```

**Additional context**

I'm using some cheap clone from Aliexpress (Piswords).

iceman1001, 2019-12-31:
the mfu_magic script works with magic ntag 21* cards as clearly stated in the help text.  Are you using a magic NTAG 21* card?
```
v1.1.1
This script enables easy programming of a MAGIC NTAG 21* card
```

pp614, 2020-01-01:
Ok, sorry, didn't know those where different types. My bad.

closed 2020-01-01 by pp614