RfidResearchGroup/proxmark3 #511: Compile Error

7h3cr0w, 2020-01-05:
Hi, I tried to compile the project under Kali and Debian. Every time i got the same error:

$ make all

===================================================================
Platform name:     Proxmark3 rdv4
PLATFORM:          PM3RDV4
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
[-] CC util.c
util.c: In function ‘strmcopy’:
util.c:925:9: error: ‘strncat’ specified bound depends on the length of the source argument [-Werror=stringop-overflow=]
  925 |         strncat(dest, src, len);
      |         ^~~~~~~~~~~~~~~~~~~~~~~
util.c:922:15: note: length computed here
  922 |     int len = strlen(src) + 1;
      |               ^~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:437: obj/util.o] Error 1
make: *** [Makefile:98: client/all] Error 2

iceman1001, 2020-01-05:
yeah... that bugger... I sort it out...

iceman1001, 2020-01-05:
Test again. I think I sorted it out.

closed 2020-01-05 by iceman1001