RfidResearchGroup/proxmark3 #512: after flashing the pm the proxmark did not come back "Waiting for Proxmark3 to appear...

PaulaPaul750, 2020-01-06:
Hello,

i tried to flash the proxmark with no success.
my steps

- deinstallation of the modemmanager
- running pm3-flash-all

failed with hint: flash bootloder and fullimage separatly

doing 
roxmark3 /dev/ttyACM0 --flash --unlock-bootloader --image bootrom.elf

finsihed successfully

doing
pm3-flash -b /tmp/my-bootrom.elf /tmp/my-fullimage.elf

works successfully.

trying to access proxmark3 /dev/ttyACM0 fails.
checking :

running sudo dmesg | grep -i usb
[ 2742.797787] usb 1-4: device not accepting address 50, error -62
[ 2742.797849] usb usb1-port4: unable to enumerate USB device
[ 2751.198205] usb 1-4: new full-speed USB device number 51 using xhci_hcd

flashing the full-image (pressing the button before connection, plug in, flash with
pm3-flash-fullimage

running successfull to finish, but device still not resonding:
[ 2721.444644] usb 1-4: new full-speed USB device number 49 using xhci_hcd
[ 2732.045208] usb 1-4: device not accepting address 49, error -62
[ 2732.165241] usb 1-4: new full-speed USB device number 50 using xhci_hcd
[ 2742.797787] usb 1-4: device not accepting address 50, error -62
[ 2742.797849] usb usb1-port4: unable to enumerate USB device
[ 2751.198205] usb 1-4: new full-speed USB device number 51 using xhci_hcd

any ideas how to fix that?

best

paul

iceman1001, 2020-01-06:
Usually  ```dmesg | tail``` gives which new serial port your device got.

How about you use  ```./pm3```  script when trying to start the client?

codeexchange, 2020-01-08:
After flashing with  pre-release v4.0 installed via home-brew my PM3  also does not appear as USB device (in USB prober / System Profiler) on my Mac (Mojave).
When I press the button on power on it appears as serial port as /dev/tty.usbmodemiceman1

iceman1001, 2020-01-08:
@codeexchange   ...OSX and Pm3 Easy.    

You must compile with PLATFORM PM3OTHER,  in order to make it work on Pm3 Easy.
Users of OSX / Homebrew seems to have issues.  Usually sorted with cloning the repo and compile yourself. 

https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Mac-OS-X-Homebrew-Installation-Instructions.md
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md#platform

codeexchange, 2020-01-08:
I followed the homebrew instructions: 

```
export HOMEBREW_PROXMARK3_PLATFORM=PM3OTHER
brew install --with-blueshark --HEAD proxmark3
```

iceman1001, 2020-01-08:
You shouldn't install  **--with-blueshark**  ....