RfidResearchGroup/proxmark3 #516: LF simulation (all modulations) not working after 2020-01-01

danshuk, 2020-01-11:
**Describe the bug**
LF simulation is not working with FPGA image built 2020-01-01 (i.e. after commit [05f2fbc](https://github.com/RfidResearchGroup/proxmark3/commit/05f2fbc75a5e89f7f45b7f70e90852f82f6968ae)).

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout an earlier commit, e.g. [2b76c8b](https://github.com/RfidResearchGroup/proxmark3/commit/2b76c8b669de9722176ad180a9a1f630859cbcd7) `git checkout 2b76c8b669de9722176ad180a9a1f630859cbcd7`
2. `make clean && make`
3. `./pm3-flash-fullimage`
4. `./pm3`
5. Run any LF simulation e.g. `lf viking sim 1A337`
6. Observe with LF reader / second PM3, etc. that sim is working.
7. Checkout commit [05f2fbc](https://github.com/RfidResearchGroup/proxmark3/commit/05f2fbc75a5e89f7f45b7f70e90852f82f6968ae) `git checkout 05f2fbc75a5e89f7f45b7f70e90852f82f6968ae`
8. Repeat steps 2 - 5.
9. Observe with LF reader / second PM3, etc. that sim is no longer working. Button does nothing; only way to abort is to disconnect device.
 
**Expected behavior**
Working LF simulation.

iceman1001, 2020-01-11:
yup,  I noticed yesterday aswell.  Good catch.

temskiy, 2020-01-11:
Latest commit with working LF simulation is 21ffdec

iceman1001, 2020-01-11:
Pull latest and try again.
This commit should fixed it  https://github.com/RfidResearchGroup/proxmark3/commit/621eb129764b90ac464d91b134f06291b01b082f

temskiy, 2020-01-12:
Yes! It`s work again. At 621eb12 and at latest too. Thanks.

closed 2020-01-12 by iceman1001