RfidResearchGroup/proxmark3 #520: Android Termux crash running 'hf mf autopwn'

h4waii, 2020-01-13:
**Describe the bug**
Running `hf mf autopwn` on an Android device with Termux causes a crash due to:

> fortify pthread_mutex_lock called on a destroyed mutex

I collected a coredump through gdb:
```
Core was generated by `/client/proxmark3'.
Program terminated with signal SIGABRT, Aborted.
#0  0x0000007fb7ae1f38 in abort ()
   from /system/lib64/libc.so
[Current thread is 1 (LWP 23433)]
(gdb) bt
#0  0x0000007fb7ae1f38 in abort ()
   from /system/lib64/libc.so
#1  0x0000007fb7b442f4 in __fortify_fatal(char const*, ...) () from /system/lib64/libc.so
#2  0x0000007fb7b438cc in pthread_mutex_lock ()
   from /system/lib64/libc.so
#3  0x00000055555fb7c4 in free_statelist_cache ()
    at cmdhfmfhard.c:1624
#4  0x00000055555f8ef4 in mfnestedhard (
    blockNo=<optimized out>, keyType=<optimized out>,
    key=<optimized out>, trgBlockNo=<optimized out>,
    trgKeyType=<optimized out>, trgkey=<optimized out>,
    nonce_file_read=<optimized out>,
    nonce_file_write=false, slow=<optimized out>,
    tests=0, foundkey=0x7fffffe620, filename=0x0)
    at cmdhfmfhard.c:2382
#5  0x00000055555e4ee8 in CmdHF14AMfAutoPWN (
    Cmd=<optimized out>) at cmdhfmf.c:2201
#6  0x00000055556372f4 in CmdsParse (
    Commands=0x5555723720 <CommandTable>,
    Cmd=0x7fb722c996 "autopwn") at cmdparser.c:222
#7  0x00000055556372f4 in CmdsParse (
    Commands=0x55557229d8 <CommandTable>,
    Cmd=0x7fb722c993 "mf autopwn") at cmdparser.c:222
#8  0x00000055556372f4 in CmdsParse (
    Commands=0x5555726de0 <CommandTable>,
    Cmd=0x7fb722c990 "hf mf autopwn") at cmdparser.c:222
--Type <RET> for more, q to quit, c to continue without paging--
#9  0x000000555557e17c in main_loop (
    script_cmds_file=<optimized out>, script_cmd=0x0,
    stayInCommandLoop=<optimized out>) at proxmark3.c:265
#10 0x000000555557f0d0 in main (argc=<optimized out>,
    argv=<optimized out>) at proxmark3.c:843
```

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to PM3 (I'm using an "Easy" HW version)
2. Issue `hf mf autopwn` on a supported card
3. Crash

**Expected behavior**
Expect process to complete as it does on x86_64

**Screenshots**
Not applicable

**Desktop (please complete the following information):**
 - OS: Android 9 / LineageOS 16.1
 - client: RRG/Iceman compiled with GCC 9.2.1 20190827 (Red Hat 9.2.1-1) OS:Linux ARCH:x86_64 
- bootrom: RRG/Iceman/master/e4bd3544 2020-01-12 12:57:34
- os: RRG/Iceman/master/e4bd3544 2020-01-12 12:57:4 compiled with GCC 9.2.0
- LF image built for 2s30vq100 on 2020-01-12 at  0:27:43
- HF image built for 2s30vq100 on 2020-01-12 at  0:27:54          
- uC: AT91SAM7S256 Rev C          
- Embedded Processor: ARM7TDMI          
- Nonvolatile Program Memory Size: 256K bytes, Used: 242996 bytes (93%) Free: 19148 bytes ( 7%)          
- Second Nonvolatile Program Memory Size: None          
- Internal SRAM Size: 64K bytes         
- Architecture Identifier: AT91SAM7Sxx Series          
- Nonvolatile Program Memory Type: Embedded Flash Memory     

**Additional context**
Please advise if you'd like the coredump file only if you're unable to reproduce.

iceman1001, 2020-01-13:
Interesting,
the free seems to be inside **free_statelist_cache()**  which resides inside hardnested.
This means you should get the same bug when running
**hf mf hardnested**
Can you confirm that?

h4waii, 2020-01-13:
Yes, once a key is "found" through the `hardnested` attack is when it crashes after its printed out.

iceman1001, 2020-02-22:
Did you sort this one out?

h4waii, 2020-02-22:
I think this is a platform issue, if someone with a device other than one running LineageOS 16.0 can test perhaps to confirm...

I'm compiling a new build for my phone now with some modifications and will advise.

iceman1001, 2020-02-23:
Seem like lineageOS is more careful about its mutex_lock checks.
First time finding a key,  should free all mutexs used in hardnested.
Starting the second time, should reinit all mutexs.. but here it doesn't...

iceman1001, 2020-02-23:
I found a bad spot,  "generate candidates" calls init() which uses that mutex, before initializating the mutex.  This would trigger on the second run.  Test it.

closed 2020-02-23 by iceman1001

h4waii, 2020-02-23:
hardnested still crashes calling a destroyed mutex.

iceman1001, 2020-02-23:
so there are more places to fix...  well,  that should give you an indication where to look. I have no time for this.

h4waii, 2020-02-23:
I will give this a shot, thanks for the pointer and tips.

iceman1001, 2020-02-24:
that PR breaks hardnested when running more than once in a row on Kali/Ubuntu.

h4waii, 2020-02-25:
Manually run, then run again?

iceman1001, 2020-02-25:
after reverting,  the autopwn and hardnested runs several times in a row without hickups.
I suppose termux still has problems with the mutex.

h4waii, 2020-02-25:
Ran fine concurrently on my Fedora 31 install, will need further testing.

iceman1001, 2020-02-27:
Try pulling latest and test again.  I think I another bug out of hardnested code

h4waii, 2020-02-28:
hardnested runs to completion now.

commit 8b10107c (2020-02-23): fix: #520 - calling init before initialize mutexs is bad
commit d9e07341 (2020-02-24): fix: #520 - free mutex after work is done

Why not just free after key found or not found?
commit fc84a285 (2020-02-24): Merge pull request #573 from h4waii/patch-3

fix: #520 - free mutex after work is done