RfidResearchGroup/proxmark3 #523: indala E8 card isnt being recognized

underfl0w, 2020-01-14:
**Describe the bug**
I've gotten my hands on access card of an office building which is an Indala E8 type tag. I was able to read and clone the tag using the official proxmark3 master but not with RNG/Iceman

`lf indala read` does not give the expected results.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone a similar tag to T5577 
`lf indala clone -l -r 80000002671D7561706500F1F7F2AF5B8B5ED74F256693BF442FF350`
2. Try and read the tag
3. Tag will not be read properly 

**Expected behavior**
I would expect to be able to read the tag similar to how "official" would read the tag

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Manjaro Linux 
 - inside proxmark3 client run the following commands and paste the output here.

```
 [ Proxmark3 RFID instrument ]           

 [ CLIENT ]           
  client: RRG/Iceman          
  compiled with GCC 9.2.0 OS:Linux ARCH:x86_64          

 [ PROXMARK3 ]           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/7c913265 2020-01-14 13:10:44
       os: RRG/Iceman/master/7c913265 2020-01-14 13:10:54
  compiled with GCC 9.2.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-01-12 at 15:31: 2
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 247536 bytes (47%) Free: 276752 bytes (53%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

          
[usb] pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing           
#db#   tracing ................1          
#db#   traceLen ...............0          
#db# Currently loaded FPGA image           
#db#   mode.................... LF image built for 2s30vq100 on 2020-01-12 at 15:31: 2          
#db# LF Sampling config           
#db#   [q] divisor.............95 ( 125.00 kHz )          
#db#   [b] bits per sample.....8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db#   [s] samples to skip.....0           
#db# LF T55XX config           
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]          
#db#            mode            |start|write|write|write| read|write|write          
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3          
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------          
#db# fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |           
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |           
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |           
#db#    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |           
#db#           
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......331264          
#db#   Transfer Speed PM3 -> Client = 662528 bytes/s          
#db# Various           
#db#   DBGLEVEL................1          
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308          
#db#   Slow clock..............30769 Hz          
#db# Installed StandAlone Mode           
#db#   HF Mifare sniff/simulation - (Craig Young)          
[usb] pm3 --> data tune
          
[=] Measuring antenna characteristics, please wait...
          
..
          
[+] LF antenna: 38,10 V - 125,00 kHz          
[+] LF antenna: 30,59 V - 134,83 kHz          
[+] LF optimal: 38,52 V - 126,32 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 30,49 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 88 is 134,83 kHz, 95 is 125,00 kHz.
```
`

iceman1001, 2020-01-14:
Your lf device timings indicate you are on a non-rdv4 device.  
Your clone didn't return success,  indicating  client couldnt verify the written data.
Have you tried looking at your t55xx data and see if it matches the blocks that was suppose to be written?

underfl0w, 2020-01-14:
It's a proxmark3 easy with the firmware PLATFORM=PM3OTHER set. 

You're right, the writing to the chip did not succeed. It did however work when I used the example command in the documentation 
`lf indala clone -l -r 80000001b23523a6c2e31eba3cbee4afb3c6ad1fcf649393928c14e5`

![image](https://user-images.githubusercontent.com/16727235/72346788-4ca96c00-36df-11ea-8f78-a4c901164378.png

But it still fails too read the original keyfob or clone it to the t55xx. 

When using the official proxmark3 firmware I was able to read the real keyfob and clone it.  
`lf indala clone 80000002671d756170651ef1f782af5a8b5ad63f25559fbf2a2e1b53 l` 

[image]

Now after reflashing RNG/Iceman it gives me this when using `lf search`
[image]

iceman1001, 2020-01-14:
Like I said,  can you look at the t55xx data and see what got written?

underfl0w, 2020-01-14:
`[usb] pm3 --> lf t55xx dump
[+] Reading Page 0:          
[+] blk | hex data | binary                           | ascii          
[+] ----+----------+----------------------------------+-------          
[+]  00 | 000820E0 | 00000000000010000010000011100000 | .. .          
[+]  01 | 00C000C0 | 00000000110000000000000011000000 | ....          
[+]  02 | 671D7561 | 01100111000111010111010101100001 | g.ua          
[+]  03 | 70651EF1 | 01110000011001010001111011110001 | pe..          
[+]  04 | F782AF5A | 11110111100000101010111101011010 | ...Z          
[+]  05 | 8B5AD63F | 10001011010110101101011000111111 | .Z.?          
[+]  06 | 25559FBF | 00100101010101011001111110111111 | %U..          
[+]  07 | 2A2E1B53 | 00101010001011100001101101010011 | *..S          
[+] Reading Page 1:          
[+] blk | hex data | binary                           | ascii          
[+] ----+----------+----------------------------------+-------          
[+]  00 | 000820E0 | 00000000000010000010000011100000 | .. .          
[+]  01 | E03900D0 | 11100000001110010000000011010000 | .9..          
[+]  02 | 7920F272 | 01111001001000001111001001110010 | y .r          
[+]  03 | 00A00003 | 00000000101000000000000000000011 | ....         `

iceman1001, 2020-01-14:
hm,  looks the same.
Only thing I came come up with now is the config block.  
```
lf indala clone on offical
lf t55xx detect 
lf indala clone on iceman
lf t55xx detect
```
And see if the config blocks matchs.

underfl0w, 2020-01-14:
So the difference is that 
[image]
And `lf t55xx detect ` iceman detects the same 
[image]

Things do change a lot after cloning it when on Iceman. 
[image]

Official does recognize the clone that is made on Iceman.

iceman1001, 2020-01-14:
Pull latest and try again.
the preamble detection for 224bits is less strict now.
https://github.com/RfidResearchGroup/proxmark3/commit/32fdde3d00924abf41a3620d5c85ce28e924679a

underfl0w, 2020-01-14:
I've verified that I'm able to read the tag now! 
Thanks a ton for fixing this so fast. 

It has made me want to dive into actually start developing instead of just using your superb software. 

[image]

closed 2020-01-14 by iceman1001