RfidResearchGroup/proxmark3 #525: Issue w/ LF_SAMYRUN LEDs in Standalone Mode

jsarkisian, 2020-01-15:
After flashing my new Proxmark 3 RDV 4.01, adding the BTADDON to the makefile, and using the LF_SAMYRUN standalone mode, I am not getting the expected LEDs when compared to the chart found here:

https://twitter.com/waveguyd/status/1051546698500067328

Instead, this is what is happening:

1. Turn on device, red LED briefly lights up at D and then goes out.

2. Hold button for 2-3 seconds, lights all light up in sequence back and forth, then land on D again briefly, then all go out.

3. Hold button again for 2 seconds, A begins to light up, then D begins to blink. It is now in reading mode and will capture a LF badge. Once it does, A blinks, then all lights go out.

4. Hold button for 1 second, C lights up and it is now in replay mode for the LF card it just captured. This also works and can open the associated doors.

5. Press button again quickly, all lights go out, no longer in replay mode.

6. Hold button for 2 seconds, both A and D flicker twice and go out. **I have zero clue what this does.**

7. Hold button for 2 seconds again, B lights up, and D begins to blink. I assume this is now using the other memory bank to again read a card.

8. Tap a card, it reads it, B blinks once and all lights go out.

9. Hold button for 1 second, C lights up and it is now in replay mode and works.

10. Tap button again, B and C blink and then all lights go out.

11. Hold button for 2 seconds, A and D blink once, then B blinks once, then all go out.

12. Hold button for 2 seconds, and we start over again at step 3.

My questions are:

1. Is this normal, but a new sequence based on latest flash?
2. What other functionality am I missing given that the above steps do not match the linked chart?
3. Is there a repo, wiki, how-to, etc. that explains exactly what I'm asking? I've been googling and all over GitHub, rfidresearchgroup, etc. and can't find what I need.

Thank you!

iceman1001, 2020-01-16:
I think I changed the led output to be more intuitive.  
As seen in the commit message.
https://github.com/RfidResearchGroup/proxmark3/commit/4eab354f448a87c2426a707639af5c1d1a97b2b8#diff-761916f224d3fad88c5cfc51833aeb5f

jsarkisian, 2020-01-16:
Fair enough, just having a hard time dealing with LEDs being a bit erratic. Sometimes it will boot up, you can enter reading mode, etc. no problem and the LEDs will be straight forward. Other times you turn it on, try and enter reading mode, and no red LEDs light up to the point where I have wondered if it is broken. I'll have to quick press the button a bunch of times, hold it in for varying amounts of time, etc. and then randomly the LEDs will begin to function properly again.

Any idea on why so erratic? Does it need time after powering on before you press anything? Sorry, maybe dumb questions, but I can find zero info on the problems I'm having.

iceman1001, 2020-01-17:
No idea.  If you think you can make a better led feedback user experience,  feel free to contribute.

jsarkisian, 2020-01-17:
Wasn't meant to offend. Was just hoping for a little more insight on the code you changed. While I can't program in C, the commit comments really aren't clear. Thanks

closed 2020-01-17 by jsarkisian

iceman1001, 2020-01-17:
non taken. 
Start with looking at the code and map out the leds for different scenarioes.