RfidResearchGroup/proxmark3 #526: firmware images not present

hornyd, 2020-01-16:
Hi,

We have rebuild the project on a new computer and follow the updated procedure
Using ProxSpace and this pm3

After running runme.bat and the make clean && make all, eveything is present except the images .elf files to update the Proxmark firmware.

Is any new procedure or is the process different?

When we try to run, it mentioned we have to update the firmware

iceman1001, 2020-01-16:
Too little information.  I have no clue what you actually done and what was the output.

Did you follow the guides?
- https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Windows-Installation-Instructions.md
- https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/0_Compilation-Instructions.md
- https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/1_Validation.md
- https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/2_Configuration-and-Verification.md

hornyd, 2020-01-16:
Sorry, after clearing the computer and doing it from scratch it got all files. Not sure what was the problem. We will close this issue.

closed 2020-01-16 by hornyd