RfidResearchGroup/proxmark3 #527: Improving LF Signal Demodulation

Silipwn, 2020-01-17:
Hi guys! 
I would like to work on improving the LF signal demodulation, was just wondering what all should I look into before starting to work on the same. I'm currently reading the `common/lfdemod.c`. Any help or things to look at would be helpful. Thanks in advance. :)

iceman1001, 2020-01-17:
For client side handling of data,  these two files are often involved.
- `common/lfdemod.c`
- `client/cmddata.c`

The lua scripts for testing lf demodulation is found here
- `https://github.com/RfidResearchGroup/proxmark3/blob/master/client/luascripts/test_t55x7.lua`
- `https://github.com/RfidResearchGroup/proxmark3/blob/master/client/luascripts/test_t55x7_ask.lua`
etc

The first script gives a good indication on which demods needs improving.

Silipwn, 2020-01-17:
Thanks for the update.

iceman1001, 2020-04-25:
@Silipwn   time to close?

Silipwn, 2020-04-25:
Sure, I will close.

closed 2020-04-25 by Silipwn