RfidResearchGroup/proxmark3 #529: [Build] /usr/lib64/gcc/arm-none-eabi/9/ld: error: no memory region specified for loadable section `.note.gnu.build-id'

hsanjuan, 2020-01-18:
When building, gcc fails with:

`/usr/lib64/gcc/arm-none-eabi/9/ld: error: no memory region specified for loadable section `.note.gnu.build-id'`. This happens in several places (like the `obj/fullimage.elf` object).

Google suggested I could fix this by adding `-Wl,--build-id=none,...` and that seems to work, but I'm not sure why the error happens in the first place or the consequences of the fix.

hsanjuan, 2020-01-18:
I must add: latest `master`, on Linux (openSUSE Tumbleweed), with `cross-arm-none-gcc9-9.2.1+r279103`

iceman1001, 2020-01-19:
No idea,  openSUSE and its arm-non-eabi compiler,   sounds like @doegox might know something

doegox, 2020-01-20:
ok I verified it in a docker image.
The fix you proposed seems indeed to be the right thing to do.
I also added installation instructions for opnSUSE in https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md
Thank you @hsanjuan !

closed 2020-01-20 by doegox

hsanjuan, 2020-01-21:
Thanks so much @doegox, I very much appreciate it.

commit fc007554 (2020-01-20): Fix compilation under openSUSE, thanks @hsanjuan. Fixes #529