RfidResearchGroup/proxmark3 #536: Default Hitag2 Reader password fails

Danyc0, 2020-01-22:
**Describe the bug**
The client help says that the default password is 4D494B52, but this doesn't seem to work. However if you manually specify this as the password, it works.

**To Reproduce**
Steps to reproduce the behavior:
1. Put tag with password 4D494B52 on pm3
2. Run: lf hitag reader 21
3. Observe that read fails
4. Run: lf hitag reader 21 4D494B52
5. Observe that read succeeds

**Expected behavior**
"lf hitag reader 21" should successfully read a tag with password 4D494B52 by default, without having to specify the password (as in the documentation)

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - hw version: 
[image]

 - hw status: 
[image]

 - data tune: 
[image]

iceman1001, 2020-01-22:
This is all a multi step refactoring. 
So it will take some time before it all is functional.

Danyc0, 2020-01-22:
If you need a hand with anything please let me know. I tried to find where in the code the default is set, but my knowledge of the client code base let me down.

commit fc1cede7 (2020-01-22): fix: #536
closed 2020-01-22 by iceman1001