RfidResearchGroup/proxmark3 #540: Drawbacks of choosing this repo over the original?

lucidBrot, 2020-01-28:
Sorry if this has been covered, but I haven't been able to find a good overview on when I should use which repository.

When should I resort to which of these?  
* https://github.com/Proxmark/proxmark3
* https://github.com/RfidResearchGroup/proxmark3
* https://github.com/iceman1001/proxmark3 (My guess: never, not even on older devices)

I see some advantages listed in the old iceman fork readme. But what are the disadvantages?

iceman1001, 2020-01-28:
When to resort?   Odd question,  normally you have a purpose with your proxmark3. Then you try to find a solution to your purpose,  That means different firmware / client.  You try until you find a solution that fits your purpose.  

There are no "mac vs pc' fanboi idea behind.  Proxmark3 was design to easily swap firmware.

lucidBrot, 2020-01-28:
Oh okay, so the mentality is not "flash firmware once, then never touch firmware again".

What I'm looking for is some overview that helps me choose. Or rather - when would I prefer the original proxmark firmware?

iceman1001, 2020-01-28:
Exactly,  you will most likely flash firmware very often.  You will want the latest source since it fixes stuff in one way or another.

As mentioned, you have a problem you wanna use your proxmark3 to solve or help.
Now then you try offical repo.  DId it solve it?  if yes, you are happy.
Didn't solve it,  then you try another repo, usually RRG/Iceman repo.   Did it solve it?  if yes, then you are happy.   Didn't solve it,  try old iceman fork,  ...  and the story continues.
You start searching for more obscure firmware versions,  because you are curious etc,  you search the forum,  on how to solve your problem.

Now the closest to what is different in a repo  is the CHANGELOG.md file.   
I tend to try this repo first on my rdv4 devices,  and if that doesn't help I swap to offical repo.
If that doesn't help, I tend to to notice what is not working and adding it to a gigantic todo list.
100 items at the moment.  

Enjoy!

closed 2020-01-29 by iceman1001

lucidBrot, 2020-01-29:
I see. Thank you very much for your time!

I will likely try this fork then, because of the improved legic support.
A list with the most important differences would have been nice, but I can see how that can be too much effort to be worth it.

iceman1001, 2020-01-29:
If you want to create a document which keeps track of the most important differences,  then be my guest.  Feel free to contribute.