RfidResearchGroup/proxmark3 #541: lf t5 det fails after hitag changes

swg0101, 2020-01-28:
**Describe the bug**
Commit d956e8c2a30d4f74f270da1d84b5ff71d4246757 was the last commit that lf t5 det works properly. After the hitag changes are added, lf t5 det always fail with "Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'". Because of that, a lot of other T55xx commands break complaining about block 0 not being able to be read.

**To Reproduce**
Steps to reproduce the behavior:
1) Compile on commit 4154f3dd1ac1e0ae6edf06f6f00a803ca34f7544.
2) Flash PM3.
3) Run lf t5 det on a valid tag.

**Expected behavior**
Modulation should be detected when a password is not used.

iceman1001, 2020-01-29:
Aha,
I suspected there was an issue,  been playing with hitag.

try pulling last commit and test again  https://github.com/RfidResearchGroup/proxmark3/commit/04bca3cdb7b0ef7f6085f39f27ff21753e0bc193

swg0101, 2020-01-29:
`u0_a134@localhost ~> pmb 04bca3c
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 18 (delta 12), reused 17 (delta 11), pack-reused 0
Unpacking objects: 100% (18/18), done.
From https://github.com/RfidResearchGroup/proxmark3
   aae8f6ae..ebc04ea2  master     -> origin/master
Previous HEAD position was d956e8c2 textual
HEAD is now at 04bca3cd fix partial data acqusitions not use bigbuff_malloc

===================================================================
Platform name:     Proxmark3 Generic target
PLATFORM:          PM3OTHER
Platform extras:   No extra selected
Included options:  LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE bootrom/all
[*] MAKE fpga_compress/all
make[1]: Nothing to be done for 'all'.
[*] MAKE armsrc/all
[-] CC version.c
[-] CC ../common_arm/usb_cdc.c
[-] CC ../common_arm/clocks.c
[-] CC bootrom.c
[-] CC fpga_version_info.c
[-] CC start.c
[-] CC iso15693.c
[-] CC ../common/iso15693tools.c
[-] CC thinfilm.c
[=] LD obj/bootrom.elf
[-] CC lfops.c
[=] GEN obj/bootrom.s19
[-] CC lfsampling.c
[-] CC pcf7931.c
[-] CC ../common/lfdemod.c
[-] CC lfadc.c
[-] CC ../common/zlib/inflate.c
[-] CC ../common/zlib/inffast.c
lfadc.c:140:6: error: conflicting types for 'lf_init'
 void lf_init(bool reader, bool simulate) {
      ^~~~~~~
In file included from lfadc.c:9:0:
lfadc.h:26:6: note: previous declaration of 'lf_init' was here
 void lf_init(bool reader);
      ^~~~~~~
make[1]: *** [../common_arm/Makefile.common:60: obj/lfadc.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:95: armsrc/all] Error 2
u0_a134@localhost ~>
`

swg0101, 2020-01-29:
> u0_a134@localhost ~> pmb 04bca3c
> remote: Enumerating objects: 18, done.
> remote: Counting objects: 100% (18/18), done.
> remote: Compressing objects: 100% (7/7), done.
> remote: Total 18 (delta 12), [...]

swg0101, 2020-01-29:
```
u0_a134@localhost ~> pmb 04bca3c
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 18 (delta 12), reused 17 (delta 11), pack-reused 0
Unpacking objects: 100% (18/18), done.
From https://github.com/RfidResearchGroup/proxmark3
   aae8f6ae..ebc04ea2  master     -> origin/master
Previous HEAD position was d956e8c2 textual
HEAD is now at 04bca3cd fix partial data acqusitions not use bigbuff_malloc

===================================================================
Platform name:     Proxmark3 Generic target
PLATFORM:          PM3OTHER
Platform extras:   No extra selected
Included options:  LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE bootrom/all
[*] MAKE fpga_compress/all
make[1]: Nothing to be done for 'all'.
[*] MAKE armsrc/all
[-] CC version.c
[-] CC ../common_arm/usb_cdc.c
[-] CC ../common_arm/clocks.c
[-] CC bootrom.c
[-] CC fpga_version_info.c
[-] CC start.c
[-] CC iso15693.c
[-] CC ../common/iso15693tools.c
[-] CC thinfilm.c
[=] LD obj/bootrom.elf
[-] CC lfops.c
[=] GEN obj/bootrom.s19
[-] CC lfsampling.c
[-] CC pcf7931.c
[-] CC ../common/lfdemod.c
[-] CC lfadc.c
[-] CC ../common/zlib/inflate.c
[-] CC ../common/zlib/inffast.c
lfadc.c:140:6: error: conflicting types for 'lf_init'
 void lf_init(bool reader, bool simulate) {
      ^~~~~~~
In file included from lfadc.c:9:0:
lfadc.h:26:6: note: previous declaration of 'lf_init' was here
 void lf_init(bool reader);
      ^~~~~~~
make[1]: *** [../common_arm/Makefile.common:60: obj/lfadc.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:95: armsrc/all] Error 2
u0_a134@localhost ~>

```

iceman1001, 2020-01-29:
yeah, pull latest

swg0101, 2020-01-29:
New error:

```
u0_a134@localhost ~> pmb
Previous HEAD position was 04bca3cd fix partial data acqusitions not use bigbuff_malloc
Switched to branch 'master'
Your branch is behind 'origin/master' by 4 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
Updating aae8f6ae..ebc04ea2
Fast-forward
 armsrc/hitag2.c     | 416 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++-------------------------------------------------------------------
 armsrc/lfadc.c      |  40 +++++++++++----
 armsrc/lfadc.h      |   3 +-
 armsrc/lfsampling.c |  30 ++++++-----
 armsrc/lfsampling.h |   1 +
 fpga/lo_adc.v       |   7 +--
 6 files changed, 284 insertions(+), 213 deletions(-)

===================================================================
Platform name:     Proxmark3 Generic target
PLATFORM:          PM3OTHER
Platform extras:   No extra selected
Included options:  LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE bootrom/all
[*] MAKE fpga_compress/all
make[1]: Nothing to be done for 'all'.
[*] MAKE armsrc/all
make[1]: Nothing to be done for 'all'.
[-] CC lfops.c
[-] CC lfsampling.c
[-] CC pcf7931.c
[-] CC lfadc.c
[-] CC ../common/zlib/inffast.c
[-] CC ../common/zlib/inftrees.c
[-] CC ../common/zlib/adler32.c
[-] CC ../common/zlib/zutil.c
[-] CC legicrf.c
[-] CC legicrfsim.c
[-] CC ../common/legic_prng.c
[-] CC hitag2_crypto.c
[-] CC hitag2.c
[-] CC hitagS.c
[-] CC appmain.c
[-] CC printf.c
[-] CC dbprint.c
[-] CC ../common/commonutil.c
[-] CC util.c
[-] CC string.c
[-] CC BigBuf.c
[-] CC ticks.c
[-] CC ../common_arm/clocks.c
[-] CC hfsnoop.c
[-] CC fpgaloader.c
[-] CC iso14443a.c
[-] CC mifareutil.c
[-] CC mifarecmd.c
[-] CC epa.c
[-] CC mifaresim.c
[-] CC iso14443b.c
[-] CC ../common/crapto1/crypto1.c
[-] CC des.c
[-] CC desfire_key.c
[-] CC mifaredesfire.c
[-] CC ../common/mbedtls/aes.c
[-] CC ../common/mbedtls/platform_util.c
[-] CC iclass.c
[-] CC optimized_cipher.c
[-] CC ../common/crc.c
[-] CC ../common/crc16.c
[-] CC ../common/crc32.c
[-] CC felica.c
[-] CC ../armsrc/Standalone/lf_samyrun.c
[-] CC ../common/parity.c
[-] CC ../common_arm/usb_cdc.c
[-] CC cmd.c
[-] GEN version.c
[-] CC version.c
[=] LD obj/fullimage.stage1.elf
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: obj/lfadc.o: in function `lf_manchester_send_bytes':
lfadc.c:(.text.lf_manchester_send_bytes+0x14): undefined reference to `lf_manchester_send_bit'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: lfadc.c:(.text.lf_manchester_send_bytes+0x1a): undefined reference to `lf_manchester_send_bit'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: lfadc.c:(.text.lf_manchester_send_bytes+0x20): undefined reference to `lf_manchester_send_bit'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: lfadc.c:(.text.lf_manchester_send_bytes+0x26): undefined reference to `lf_manchester_send_bit'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: lfadc.c:(.text.lf_manchester_send_bytes+0x2c): undefined reference to `lf_manchester_send_bit'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: obj/lfadc.o:lfadc.c:(.text.lf_manchester_send_bytes+0x58): more undefined references to `lf_manchester_send_bit' follow
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:179: obj/fullimage.stage1.elf] Error 1
make: *** [Makefile:95: armsrc/all] Error 2

```

iceman1001, 2020-01-29:
pull again.  Strange, on ubuntu I don't get that fault.  Which os/gcc version do you have?

swg0101, 2020-01-29:
Ubuntu 19.04 on Android via gcc 8.3.0. Client is built via clang on Termux.
Latest builds fine, although same issue with detects not working properly.

iceman1001, 2020-01-29:
I have remove some lines in order to focus on the important parts. **lf t55xx detect** works just fine for me.
```
[usb] pm3 --> hw stat
#db# Memory
...
#db# Currently loaded FPGA image
#db#   mode.................... LF image built for 2s30vq100 on 2020-01-28 at 22:15:39
...
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125.00 kHz )
#db#   [b] bits per sample.....8
#db#   [d] decimation..........1
#db#   [a] averaging...........No
#db#   [t] trigger threshold...0
#db#   [s] samples to skip.....0
#db# LF T55XX config
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
#db#            mode            |start|write|write|write| read|write|write
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
#db#     long leading reference |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
#db#               leading zero |  29 |  17 |  15 |  40 |  15 | N/A | N/A |
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
#db#

 [ Proxmark3 RFID instrument ]
 [ CLIENT ]
  client: RRG/Iceman
  compiled with GCC 7.4.0 OS:Linux ARCH:x86_64
 [ PROXMARK3 RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK3 RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/master/cf651453-dirty-unclean 2020-01-25 13:51:56
       os: RRG/Iceman/master/d6f552e8-dirty-unclean 2020-01-29 04:33:11
  compiled with GCC 6.3.1 20170620

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-01-28 at 22:15:39
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16

[usb] pm3 --> lf t55 det
    Chip Type      : T55x7
    Modulation     : ASK
    Bit Rate       : 2 - RF/32
    Inverted       : No
    Offset         : 32
    Seq. Term.     : Yes
    Block0         : 0x000880E0
    Downlink Mode  : default/fixed bit length
    Password Set   : No
```

iceman1001, 2020-01-29:
PM3OTHER,   non-rdv4 device,  well,  check distance between tag / reader, position,  the correct timings,
otherwise no much I do.

swg0101, 2020-01-29:
It seems like if I try about 10 times it may work about 30-40% of the time even without moving the tag. Something seems to be quite odd here since I haven't had that much troubles with detects before unless it was password protected (which would be expected)...

iceman1001, 2020-01-29:
What is your timings, from **hw status**

swg0101, 2020-01-29:
```
[usb] pm3 --> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............0
#db# Currently loaded FPGA image
#db#   mode.................... HF image built for 2s30vq100 on 2020-01-12 at 15:31:16
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125.00 kHz )
#db#   [b] bits per sample.....8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db#   [s] samples to skip.....0
#db# LF T55XX config
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
#db#            mode            |start|write|write|write| read|write|write
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------
#db# fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
#db#    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
#db#
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............500ms
#db#   Bytes transferred.......259584
#db#   Transfer Speed PM3 -> Client = 519168 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db#   Slow clock..............29537 Hz
#db# Installed StandAlone Mode
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)

```

swg0101, 2020-01-29:
Without moving the tag, here's how the detects look:

```
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
    Chip Type      : T55x7
    Modulation     : FSK2a
    Bit Rate       : 4 - RF/50
    Inverted       : Yes
    Offset         : 33
    Seq. Term.     : No
    Block0         : 0x00107060
    Downlink Mode  : default/fixed bit length
    Password Set   : No

[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det
    Chip Type      : T55x7
    Modulation     : FSK2a
    Bit Rate       : 4 - RF/50
    Inverted       : Yes
    Offset         : 33
    Seq. Term.     : No
    Block0         : 0x00107060
    Downlink Mode  : default/fixed bit length
    Password Set   : No

```

iceman1001, 2020-01-29:
normally you look for the sweatspot in pos/distance between tag and antenna and remember it.
You don't lay the card flat and expect it to work....  Not the proxmark way

swg0101, 2020-01-29:
That's how I was doing it, but compared to the commit I was referring to, I cannot get the detects to work most of the times regardless of positioning, distance, etc. Like I said, even when I am not moving the tag, sometimes the detects itself would fail. This is with FSK2a so detects may be a bit more difficult there (but it never was an issue previously).

iceman1001, 2020-01-29:
feel free to find a solution in the fsk2a.  I don't have time.

closed 2020-02-27 by swg0101