RfidResearchGroup/proxmark3 #543: Blueshark won't connect

PsychoBitchy, 2020-01-29:
I recently installed the RRG/blueshark hat on my proxmark3 RDV4.0
I compiled the last firmware with PLATFORM_EXTRAS=BTADDON
All seems to work flawlessly, except the RRG/Rfid Tools android app.
The app connects to blueshark ( message: " communication normall " + still blue led )
I then get a message about connection issue.
If I try to connect via UART with " serial bluetooth Terminal " android app it echoes " connected and the blue led stands still so I'm assuming the UART protocol is fonctional.
How can I troubleshoot the connection further?
thanks.

My PM3 RDV4 Hws:
```
[usb] pm3 --> hw tune
          
[=] Measuring antenna characteristics, please wait...
          
[=] You can cancel this operation by pressing the pm3 button          
..
          
[+] LF antenna: 69.96 V - 125.00 kHz          
[+] LF antenna: 33.14 V - 134.83 kHz          
[+] LF optimal: 69.96 V - 125.00 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 44.19 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 88 is 134.83 kHz, 95 is 125.00 kHz.
          
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]           

 [ CLIENT ]           
  client: RRG/Iceman          
  compiled with GCC 7.4.0 OS:Linux ARCH:x86_64          

 [ PROXMARK3 RDV4 ]           
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK3 RDV4 Extras ]           
  FPC USART for BT add-on support: present           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/aae8f6ae-dirty-unclean 2020-01-28 15:54:05
       os: RRG/Iceman/master/aae8f6ae-dirty-unclean 2020-01-28 15:54:21
  compiled with GCC 6.3.1 20170620

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-01-12 at 15:31: 2
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev A          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 308087 bytes (59%) Free: 216201 bytes (41%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

```

iceman1001, 2020-01-29:
the app works with a custom fw, as mentioned in the rfid app repo.

PsychoBitchy, 2020-01-29:
Thank you, I've missed that point somehow

closed 2020-01-29 by PsychoBitchy