RfidResearchGroup/proxmark3 #550: HF_MATTYRUN Fails Filling Emulator Memory

h4waii, 2020-02-04:
**Describe the bug**
When running HF_MATTYRUN in standalone mode (but connected to interface to view debugging), keys are located and sectors read, but filling emulator memory fails with no verbose information.

**To Reproduce**
Compile with STANDALONE=HF_MATTYRUN, connect to client, run `hw standalone` and present tag with default, or available key, from array mfKeys.

**Expected behavior**
Expect card to be dumped using found keys, data to be stored in emulator and available to emulation or writing out to blank card

**Screenshots**
N/A, but huge text dump available:

```        
[usb] pm3 --> hw standalone
#db# Stand-alone mode, no computer necessary          
#db# >>  Matty mifare chk/dump/sim  a.k.a MattyRun Started  <<          
#db#    Current sector:  0, block:  3, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  1, block:  7, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  2, block: 11, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  3, block: 15, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  4, block: 19, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  5, block: 23, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  6, block: 27, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  7, block: 31, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  8, block: 35, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  9, block: 39, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 10, block: 43, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 11, block: 47, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 12, block: 51, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 13, block: 55, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 14, block: 59, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 15, block: 63, key type: A, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  0, block:  3, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  1, block:  7, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  2, block: 11, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  3, block: 15, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  4, block: 19, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  5, block: 23, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  6, block: 27, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  7, block: 31, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  8, block: 35, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector:  9, block: 39, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 10, block: 43, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 11, block: 47, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 12, block: 51, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 13, block: 55, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 14, block: 59, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    Current sector: 15, block: 63, key type: B, key count: 13           
#db#    ✓ Found valid key: [ffffffffffff]
          
#db#    ✕ There's nothing I can do without at least a one valid key, sorry!          
#db#    ✓ Found keys have been transferred to the emulator memory.          
#db#    Filling in with key A.          
#db#    ✕ Failed filling with A.          
#db#    Filling in with key B.          
#db#    ✕ Failed filling with B.          
#db#    ✕ Emulator memory could not be filled due to errors.          
```

**Desktop (please complete the following information):**
 - OS: N/A, Standalone Mode
```
 [ CLIENT ]           
  client: RRG/Iceman          
  compiled with GCC 9.2.1 20190827 (Red Hat 9.2.1-1) OS:Linux ARCH:x86_64          

 [ PROXMARK3 ]           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/aa0054e6 2020-02-04 14:18:04
       os: RRG/Iceman/master/aa0054e6 2020-02-04 14:18:18
  compiled with GCC 9.2.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-01-12 at 15:31: 2
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16          

 [ Hardware ]           
  --= uC: AT91SAM7S256 Rev C          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 247868 bytes (95%) Free: 14276 bytes ( 5%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          
 ```

h4waii, 2020-02-07:
Checked HF_COLIN, this also fails with the same error on the same tag with the same key.

iceman1001, 2020-02-27:
Try running with debug level 4,  to get all debug messages out.  
Somewhere inside MifareECardLoad the function seem to fail
```
hw dbg 4
````

iceman1001, 2020-02-28:
push one fix,  its only partial since I can imagine there a more bugs further down in the code.
This one sorts out the wrong message when all keys are found.

The error with ecardload ( _   ✕ Failed filling with B.    _ ) message indicates it uses the return codes wrong ,  if nothing was wrong with your card ...  since the card need to still be on the device when running

h4waii, 2020-02-28:
Seems to get further, won't be able to confirm that it's emulating correctly for a bit, but will advise once I do.

```
#db#    ✓ All keys found          
#db#    ✓ Found keys have been transferred to the emulator memory.          
#db#    Filling in with key A.          
#db#    ✕ Failed filling with A.          
#db#    Filling in with key B.          
#db#    ✕ Failed filling with B.          
#db#    ✕ Emulator memory could not be filled due to errors.          
[usb] pm3 --> 
```

iceman1001, 2020-02-28:
yup.  that would be ecardload function...

cjbrigato, 2020-03-05:
Am i anyhow concerned here ? I use my own without problem and I think I've followed the last breaking changes so it should work. No ?

iceman1001, 2020-03-05:
I think so,   could be some param or return value that is messing.   
One of these days, I would like to have statistics over how much the different standalone modes is used.

cjbrigato, 2020-03-10:
yeah so as I said, nothing to my concern here.
Mattyrun uses `MifareECardLoad` while I "vendored" one years ago I was content with and managed to continue make it working (a bit like the SAK vs CRAP case :p), and as I've tested right now, It works.

Maybe just take mine to his code ?

iceman1001, 2020-03-10:
yes.  go go go

cjbrigato, 2020-03-10:
WTB Time or TRADE for Cocacola

iceman1001, 2020-04-25:
@cjbrigato  did you ever do this one?

iceman1001, 2020-05-20:
ok,  added the eload and minor fixes to sim logic.

@msoose  Try pull latest / compile / flash and test it, please!

msoose, 2020-05-20:
My other devices are currently unavailable to confirm simulation is correct, but this runs cleanly now without errors.

I can check emulation in a few days when I have access to more gear.

msoose, 2020-05-21:
Emulation works as described. Thanks for fixing this!

closed 2020-05-21 by msoose