RfidResearchGroup/proxmark3 #551: Help debugging FPGA crash on sniffing hitag

bosb, 2020-02-04:
**Describe the bug**
use command "lf hitag sniff"
go to reader with proxmark rdv4 via USB and token hitag S
token is accepted by reader
proxmark lights 2 red leds
trying to get log: "lf hitag list"
-> nothing, communication error
I need to unplug proxmark to get it working again :-(

```
[usb] pm3 --> lf hitag sniff
#db# Starting Hitag2 sniffing
[usb] pm3 -->
[usb] pm3 --> lf hitag list
Waiting for a response from the Proxmark3...
You can cancel this operation by pressing the pm3 button
[-] Timed out while trying to download data from device
[!] timeout while waiting for reply.
[!] Communicating with Proxmark3 device failed
[=] Running in OFFLINE mode. Use "hw connect" to reconnect

[offline] pm3 --> hw connec
[=] Using UART port /dev/tty.usbmodemiceman1
[=] Communicating with PM3 over USB-CDC
```

What are my possibilities to figure out where/why it crashed?

iceman1001, 2020-02-04:
its crashing since I am reworking the hitag code.  You could go back to old versions before the hitag changes,  then it was working.

bosb, 2020-02-04:
ups, indeed, switched back to the android app compatible one from 2019-08-12

cosmo6110, 2020-02-13:
bosb
I tried this version and it worked  08e62af.

Zorro9999, 2020-07-26:
Good evening. Unfortunately, in all versions compiled for 64x-bit systems, the LF hitag sniff read mode could not be started, it allegedly starts, but when trying to read the log with the trace list command, a reconnect with the device occurs.
 [Proxmark3 RFID instrument]
``
 [CLIENT]
  client: RRG / Iceman / master / v4.9237-618-g84a49bf0 2020-07-23 22:32:27
  compiled with MinGW-w64 9.3.0 OS: Windows (64b) ARCH: x86_64

 [PROXMARK3]

 [ARM]
  bootrom: RRG / Iceman / master / v4.9237-618-g84a49bf0 2020-07-23 22:32:11
       os: RRG / Iceman / master / v4.9237-537-gac12a892 2020-06-29 13:37:15
  compiled with GCC 9.2.1 20191025 (release) [ARM / arm-9-branch revision 277599]

 [FPGA]
  LF image built for 2s30vq100 on 2020-02-22 at 12:51:14
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16

 [Hardware]
  - = uC: AT91SAM7S512 Rev B
  - = Embedded Processor: ARM7TDMI
  - = Nonvolatile Program Memory Size: 512K bytes, Used: 226976 bytes (43%) Free: 297312 bytes (57%)
  - = Second Nonvolatile Program Memory Size: None
  - = Internal SRAM Size: 64K bytes
  - = Architecture Identifier: AT91SAM7Sxx Series
  - = Nonvolatile Program Memory Type: Embedded Flash Memory
``

closed 2026-09-12 by iceman1001