RfidResearchGroup/proxmark3 #552: STANDALONE=LF_ICEHID compile error

netvader, 2020-02-04:
after `git pull && make clean && make PLATFORM=PM3RDV4 STANDALONE=LF_ICEHID PLATFORM_EXTRAS=BTADDON` i´ve got the following error

```
[-] CC ../armsrc/Standalone/lf_icehid.c
../armsrc/Standalone/lf_icehid.c: In function 'IceEM410xdemod':
../armsrc/Standalone/lf_icehid.c:108:86: error: 'sprintf' writing a terminating nul past the end of the destination [-Werror=format-overflow=]
         sprintf((char *)entry, "EM XL TAG ID: %06lx%08lx%08lx - (%05ld_%03ld_%08ld)\n",
                                                                                      ^
../armsrc/Standalone/lf_icehid.c:108:9: note: 'sprintf' output between 61 and 80 bytes into a destination of size 60
         sprintf((char *)entry, "EM XL TAG ID: %06lx%08lx%08lx - (%05ld_%03ld_%08ld)\n",
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                  hi,
                  ~~~
                  (uint32_t)(lo >> 32),
                  ~~~~~~~~~~~~~~~~~~~~~
                  (uint32_t)lo,
                  ~~~~~~~~~~~~~
                  (uint32_t)(lo & 0xFFFF),
                  ~~~~~~~~~~~~~~~~~~~~~~~~
                  (uint32_t)((lo >> 16LL) & 0xFF),
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                  (uint32_t)(lo & 0xFFFFFF));
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [../common_arm/Makefile.common:65: obj/lf_icehid.o] Fehler 1
make: *** [Makefile:95: armsrc/all] Fehler 2

```
**Desktop:**
 - OS: Linux  5.4.0-kali3-amd64 #1 SMP Debian 5.4.13-1kali1 (2020-01-20) x86_64 GNU/Linux
 - gcc (Debian 9.2.1-25) 9.2.1 20200123

iceman1001, 2020-02-04:
Good, I pushed a fix

netvader, 2020-02-04:
thanks! now compiles correctly

closed 2020-02-04 by netvader