RfidResearchGroup/proxmark3 #556: lf hid brute Not Incrementing Card ID

capnkrunchy, 2020-02-12:
**Describe the bug:**
The lf hid brute command does not increment card ID numbers. If given card ID of 10, only card ID 10 and 11 will be tried repeatedly.

**Steps to reproduce the behavior:**
Checkout commit e2d2e86
make clean && make all
sudo make install
pm3-flash-all
proxmark3 /dev/tty.usbmodemiceman1
lf hid brute w H10301 f 123 c 10 d 2000 v

**Terminal output:**
```
[usb] pm3 --> lf hid brute w H10301 f 123 d 2000 c 10 v
[=] Brute-forcing HID reader
[=] Press pm3-button to abort simulation or run another command
[=] Trying FC: 123; CN: 11;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 10;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 11;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 10;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 11;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 10;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 11;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 10;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 11;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 10;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 11;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 10;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
[=] Trying FC: 123; CN: 11;  Issue level: 0; OEM: 0
#db# Simulating with fcHigh: 10, fcLow: 8, clk: 50, STT: 0, n: 9600
```

iceman1001, 2020-02-12:
Feel free to fix it,  I don't time.

capnkrunchy, 2020-02-12:
Unfortunately I don't know enough about C to fully fix the issue, but I was able to get it half working by removing lines 443 through 447 from client/cmdlfhid.c. With those lines removed, it increments upwards properly but not downwards at all.

closed 2020-02-27 by capnkrunchy