RfidResearchGroup/proxmark3 #557: hf mf rdkey

mashrum, 2020-02-12:
Related with Mifare Classic, Card only attack

Often we have changed for the sector Key A and B but default access bytes
It means if we've got Key A for the sector using Nested Hardnested
we can read Key B by reading trailer block with Key A.
No need nested or hardnested. Just read.

hf mf rdkey - Read sector trailer to get unknown key if allowed
Usage:  hf mf rdkey  <sector number> <known key (12 hex symbols)>          
Sample: hf mf rdkey 1 FFFFFFFFFFFF  - Try to read sector trailer with known key
Output: 
```
|---|----------------|----------------|          
|sec|key A           |key B           |          
|---|----------------|----------------|          
|000|  ffffffffffff  |  123123123123  |          
|---|----------------|----------------| 
```

no need to bruteforce with dictionary
no need long time nested or hardnested if access bits allow to read trailer block
can be used for automation

mashrum, 2020-02-12:
can be used inside
> hf mf fchk

iceman1001, 2020-02-12:
If you take a look inside the code, you will find that we are using that trick since some years already.
No need for a special command,  since a normal read block will do the same.

closed 2020-02-12 by iceman1001