RfidResearchGroup/proxmark3 #558: RVD4.01 unable to communicate with PC&phone

anonymix007, 2020-02-13:
**Describe the bug**
Clients on Windows and Android cannot be used:
[image]
(Windows)

[image]
(Android)

**To Reproduce**
Steps to reproduce the behavior:
1. Open ProxSpace
2. Run proxmark3 com8
3. See error
or
1. Open Rfid Tools app
2. Select Proxmark 3
3. Select All Devices
4. Click PM3__RDV4.0
5. Wait and see error.

**Expected behavior**
I expected kinda work and not give me an error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10 (1607 LTSB)/ Android 10 (Oneplus 7 Pro, Root, Stock ROM/Kernel)

**Additional context**
Add any other context about the problem here.

iceman1001, 2020-02-13:
Wrong repo.  
[RFID app repo,  note about firmware](https://github.com/RfidResearchGroup/RFIDtools#proxmark3-firmware)

closed 2020-02-13 by iceman1001

anonymix007, 2020-02-13:
> Wrong repo.

And what about first screenshot with error (Unknown command:: 0x61334d50[!!] ERROR: cannot communicate with the Proxmark) ?

iceman1001, 2020-02-13:
same answer, wrong repo.  I even linked to the relevant text.

anonymix007, 2020-02-14:
After installing that FW I was able to use app, but windows client shows this:
[image]

anonymix007, 2020-02-15:
@iceman1001 Are there any instructions on building FW that can work with both Android app and client on Windows?

iceman1001, 2020-02-15:
sadly no.