RfidResearchGroup/proxmark3 #560: Remove hardcoded paths to bash, perl, python, etc. in scripts

sigwinch28, 2020-02-18:
**Describe the bug**
Several of the scripts in the repo (especially the `pm3-*` scripts and some in `tools/`) are hard-coded to use `/bin/bash` or `/usr/bin/perl`, etc. in their shebangs.

This isn't very portable between operating systems.
For example, FreeBSD and OpenBSD don't have `bash` in `/bin` by default: it's in `/usr/bin`.
Similarly, on NixOS `bash` might be in `/run/current-system/sw/bin`.

A more portable way to find these commands is to (ab)use the `env` command, like some scripts in the repo already do:
```
#!/usr/bin/env bash
```
```
#!/usr/bin/env python3
```

**To Reproduce**
Steps to reproduce the behavior:
1. Clone repo on FreeBSD, OpenBSD, or NixOS
2. Try to run `pm3` or `pm3-flash-all`
3. Depending on your shell, you will get an error like `bash: ./pm3: /bin/bash: bad interpreter: No such file or directory`

**Expected behavior**
The script should run, via bash.
If I change `#!/bin/bash` to `#!/usr/bin/env bash` in `pm3`, then it works:
```
./pm3
[=] Waiting for Proxmark3 to appear...
```

**Proposed fix**

Replace all instances of hard-coded `/bin/bash`, `/usr/bin/perl`, and `/usr/bin/python` interpreters in shebangs with equivalent invocations of `/usr/bin/env <interpreter_name>`

doegox, 2020-02-21:
sigh...
cf https://travis-ci.org/RfidResearchGroup/proxmark3/jobs/653521320

```
env: python3: No such file or directory
xorcheck test [Fail]
```

I didn't see that one coming...
Now trying again with explicit python3 install...

iceman1001, 2020-02-21:
i am pretty sure the next test "findbits.py" will also break :)

We would need to add Python3 in the travis yml..

doegox, 2020-02-21:
that's what I did in https://github.com/RfidResearchGroup/proxmark3/commit/ae57feeee6789b35addf09fab00859ab36bfb0a7
but it still fails
https://travis-ci.org/RfidResearchGroup/proxmark3/jobs/653531921
any idea?

doegox, 2020-02-21:
note that python3 was already supposedly installed in travis env:
```
Upgrading python3 formula. It is installed but not up-to-date.
==> Upgrading 1 outdated package:
python3 3.7.4_1 -> 3.7.6_1
```

doegox, 2020-02-21:
so back to the initial problem, env doesn't find python3...

sigwinch28, 2020-02-21:
:thinking: I'll have a look at the problem on a macOS VM w/ homebrew

doegox, 2020-02-21:
see https://github.com/RfidResearchGroup/proxmark3/pull/567
I'll look again later

sigwinch28, 2020-02-21:
Looks like the problem is that `python@3.8` is keg-only, so `python3` isn't being symlinked into `/usr/local/bin`:

https://travis-ci.org/RfidResearchGroup/proxmark3/jobs/653531921#L676

More info here:

 https://github.com/Homebrew/homebrew-core/issues/48368

Something along the lines of:
```
export PATH="/usr/local/opt/python@3.8/bin:$PATH"
```
in `.travis.yml` might fix this.

doegox, 2020-02-21:
Thanks, I'm testing it.
It's really weird because
without brew update we got:
```lrwxr-xr-x  1 travis  admin  24 Sep 25 01:47 /usr/local/opt/python -> ../Cellar/python/3.7.4_1
lrwxr-xr-x  1 travis  admin  24 Sep 25 01:47 /usr/local/opt/python3 -> ../Cellar/python/3.7.4_1
lrwxr-xr-x  1 travis  admin  24 Sep 25 01:47 /usr/local/opt/python@3 -> ../Cellar/python/3.7.4_1
```
and a /usr/local/bin/python3

after brew update:
```
lrwxr-xr-x  1 travis  admin  24 Feb 21 19:28 /usr/local/opt/python -> ../Cellar/python/3.7.6_1
lrwxr-xr-x  1 travis  admin  26 Feb 21 19:30 /usr/local/opt/python@3.8 -> ../Cellar/python@3.8/3.8.1
```
and no more /usr/local/bin/python3, probably only a /usr/local/bin/python == python 3.7.6

doegox, 2020-02-21:
ok I tested with export PATH="/usr/local/opt/python@3.8/bin:$PATH" (saw your edited comment later) but now I'm testing sth else because apparently the absence of python3 is due to a brew link error of python3 while python2 was still present:
```
==> Pouring python-3.7.6_1.mojave.bottle.tar.gz
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /usr/local
Could not symlink Frameworks/Python.framework/Headers
Target /usr/local/Frameworks/Python.framework/Headers
is a symlink belonging to python@2. You can unlink it:
  brew unlink python@2
To force the link and overwrite all conflicting files:
  brew link --overwrite python
To list all files that would be deleted:
  brew link --overwrite --dry-run python
```

So I'm testing now with a `before_install: brew link --overwrite python` to fix the links

doegox, 2020-02-21:
Closing this one, I bet https://travis-ci.org/RfidResearchGroup/proxmark3/builds/653633584 will pass unless the occasional arm toolchain download failures

PR #561 (2020-02-21): Use /usr/bin/env in shebangs
Use the `env` command in shebangs to find the correct interpreter instead of hard-coding paths.

Fixes #560
closed 2020-02-21 by doegox