RfidResearchGroup/proxmark3 #564: Convert all python scripts to python3

doegox, 2020-02-21:
- [ ] ./client/pm3_mfdread.py
- [ ] ./client/pm3_mfd2eml.py
- [ ] ./client/pm3_eml2mfd.py
- [ ] ./client/pm3_eml_mfd_test.py
- [ ] ./fpga/tests/plot_edgedetect.py
- [ ] ./tools/xorcheck.py
- [ ] ./tools/pm3_mfdread.py
- [ ] ./tools/findbits_test.py
- [ ] ./tools/pm3_mfd2eml.py
- [ ] ./tools/findbits.py
- [ ] ./tools/pm3_eml2mfd.py
- [ ] ./tools/pm3_eml_mfd_test.py

And this one can probably be fixed to python3 too:
- [ ] ./tools/pm3_mf7b_wipe.py #! /usr/bin/env python3.6

iceman1001, 2020-02-21:
@sigwinch28 
If you are up to it,  start converting to python3 would be nice.

sigwinch28, 2020-02-21:
I will see how far i get with `2to3`.

I'll catalog the dependencies that the scripts have, too: some have numpy dependencies, etc.

iceman1001, 2020-02-21:
that is a good way forward.

sigwinch28, 2020-02-21:
Dependency search:

* [x]  `./fpga/tests/plot_edgedetect.py`: `numpy`, `matplotlib`
* [x] `./tools/btaddon/hc06_factory.py`: `pySerial`?
* [x] `./tools/btaddon/hc06_console_AT.py`: `pySerial`?
* [x]  `./tools/xorcheck.py`: no external dependencies
* [x]  `./tools/pm3_mfdread.py`: `bitstring`
* [x]  `./tools/findbits_test.py`: no external dependencies, imports `findbits` locally.
* [x]  `./tools/pm3_mfd2eml.py`: no external dependencies
* [x]  `./tools/findbits.py`: no external dependencies
* [x]  `./tools/pm3_eml2mfd.py`: no external dependencies
* [x]  `./tools/pm3_eml_mfd_test.py`: no external dependencies, imports `pm3_eml2mfd` and `pm3_mfd2eml` locally
* [x]  `./tools/pm3_mf7b_wipe.py`: no external dependencies

Looks like the dependencies are:
- `fpga/tests`: `numpy` and `matplotlib`
- `tools`: `bitstring`
- `tools/btaddon`: `pySerial` (unconfirmed)

doegox, 2020-02-21:
Thanks for the list!
It is a bit overkill to require every pm3 user to install these dependencies.
These scripts are for very specific and rare use.
I prefer to catch properly the errors and guide the user.

sigwinch28, 2020-02-21:
> Thanks for the list!
> It is a bit overkill to require every pm3 user to install these dependencies.

I agree, but perhaps we could document them somehow, for example:
- adding comments to the tools which have external dependencies; or
- adding a `requirements.txt` to the relevant directories (or whatever is fashionable these days)

doegox, 2020-02-21:
requirements.txt are more for at least a directory of related tools, here we have all kind of scripts arbitrarily grouped.

See latest commits, the 4 scripts are now simply exiting with "Please install XXX module first" messages

iceman1001, 2020-02-21:
Very few compiles FPGA code.  Even less runs the tests.  :)

doegox, 2020-02-21:
lol, I even never noticed the fpga test script before today

doegox, 2020-02-21:
done, thanks for the help @sigwinch28 !

closed 2020-02-21 by doegox