RfidResearchGroup/proxmark3 #569: [idea] Tag Downloadable Releases

h4waii, 2020-02-22:
**Is your feature request related to a problem? Please describe.**

I would like to maintain a package for Termux, but it requires HTTP/S and a matching sha256sum for downloads, if Releases can be tagged and available for download I will keep the packages as up to date as possible when "meaningful" changes (feature updates, outstanding fixes, user improvements, etc) are committed.

**Describe the solution you'd like**
Add Releases with some sort of version tagging perhaps?

**Describe alternatives you've considered**
I would have to fork, add my own unofficial Releases and keep my fork up to date. This is an option if this idea isn't implemented, but I would much prefer to get source directly from the creators.

iceman1001, 2020-02-22:
Duplicate question.   https://github.com/RfidResearchGroup/proxmark3/issues/319

Short answer, yes,  once we are a level the repois in such shape it could be an release.

h4waii, 2020-02-22:
Okay, thanks for the answer. I will hack around this for the Termux package and go directly for master, if I can get this included, I'll update the Wiki as well.

Is there a preferred package description I can use?
I currently have "The Swiss Army Knife of RFID Research".

iceman1001, 2020-02-23:
"The Swiss Army Knife of RFID Research"  is a good one,  usually used to denotate Proxmark3.
You could add " - RRG/Iceman repo"  in order to seperate between offical repo and this one.

h4waii, 2020-02-23:
It's done. Waiting for it to hit the repositories to become available for all.

closed 2020-02-23 by h4waii