RfidResearchGroup/proxmark3 #570: hitag, determine adc levels to get better demodulation

cosmo6110, 2020-02-22:
wrong freq for me! it stop working on 125khz

FpgaSendCommand(FPGA_CMD_SET_DIVISOR, 95); //125Khz FpgaSendCommand(FPGA_CMD_SET_DIVISOR, LF_DIVISOR_134);

iceman1001, 2020-02-22:
Not sure I am following now.
125Khz worked for you or did 134Khz work for you?

What device do you have?   Which antenna?

cosmo6110, 2020-02-22:
I got the Proxmark3 RDV4.01 with LF Read Range (long range antenna).
After the latest update "LF_DIVISOR_134" it stop working on the 125 khz.

iceman1001, 2020-02-22:
Does your antenna have a setting for freq 125/134 and Q 14/7 ?  Take a picture of it.
If so, try 134 / 7 

otherwise try
```
hw dbg 4
data plot
lf hitag info
data sample
data ltrim 30000
```

And post a picture of the graph...

cosmo6110, 2020-02-23:
134 did the trick with my antenna on the 125khz transponder.

First picture is with  the old setting. FpgaSendCommand(FPGA_CMD_SET_DIVISOR, 95); //125Khz
125 /14
second picture is with the FpgaSendCommand(FPGA_CMD_SET_DIVISOR, LF_DIVISOR_134);
134 / 7
Thank's maybe we shall close thi issue?
[image]
[image]

iceman1001, 2020-02-23:
interesting that your tag is 125 kHz.  
I should adjust the hitag code to use the _lf config_ setting.  so we can use both 125 or 134.

closed 2020-02-23 by iceman1001