RfidResearchGroup/proxmark3 #571: pm3 doesn't recognize iClass tag

hypnoticpattern, 2020-02-22:
Hi,
I have a Proxmark3 RDV4 and I am trying to use pm3 with a HDI iClass tag. I tried with the latest release using both brew and compiling the code myself (MacOS Mojave) but the tag isn't recognized:

```
[=] Communicating with PM3 over USB-CDC

  ██████╗ ███╗   ███╗ ████╗
  ██╔══██╗████╗ ████║   ══█║
  ██████╔╝██╔████╔██║ ████╔╝
  ██╔═══╝ ██║╚██╔╝██║   ══█║     iceman@icesql.net
  ██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/
  ╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman
  compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 11.0.0 (clang-1100.0.33.12) OS:OSX ARCH:x86_64

 [ PROXMARK3 RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK3 RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/master/2804a7db 2020-02-21 15:44:02
       os: RRG/Iceman/master/2804a7db 2020-02-21 15:44:13
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-01-12 at 15:31: 2
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 285621 bytes (54%) Free: 238667 bytes (46%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
[usb] pm3 --> hf tune [Without the tag]
[+] Measuring HF antenna,click button  or pressEnter to exit
[\] 46612 mV /    46 V
[|] 46612 mV /    46 V
[+] Done.
[usb] pm3 --> hf tune [With the tag]
[+] Measuring HF antenna,click button  or pressEnter to exit
[|] 38912 mV /    38 V
[/] 38912 mV /    38 V
[+] Done.
[usb] pm3 --> hf search
[/] No known/supported 13.56 MHz tags found
[usb] pm3 --> hf search
[/] No known/supported 13.56 MHz tags found
[usb] pm3 -->
```

If I flash the official [Proxmark](https://github.com/Proxmark/proxmark3) repo I can identify the tag, dump it and do all the other common operations:

```
Prox/RFID mark3 RFID instrument
bootrom: master/v3.1.0-189-gbedae77-suspect 2020-02-20 21:38:55
os: master/v3.1.0-189-gbedae77-suspect 2020-02-20 21:38:56
fpga_lf.bit built for 2s30vq100 on 2019/11/21 at 09:02:37
fpga_hf.bit built for 2s30vq100 on 2019/11/13 at 14:52:19
SmartCard Slot: available

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 210796 bytes (40). Free: 313492 bytes (60).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> hf search

   CSN: 5f 3a xx xx xx xx 12 e0
    CC: xx xx ff ff ff ff ff ff
  Mode: Application [Locked]
Coding: ISO 14443-2 B/ISO 15693
 Crypt: Secured page, keys not locked
    RA: Read access not enabled
   Mem: 2 KBits/2 App Areas (32 * 8 bytes) [1F]
   AA1: blocks 06-12
   AA2: blocks 13-1F
 AppIA: ff ff ff ff ff ff ff ff
      : Possible iClass (legacy tag)

Valid iClass Tag (or PicoPass Tag) Found - Quiting Search
``` 

Any idea what the cause could be?

iceman1001, 2020-02-22:
There is some modern iClass cards which isn't identified.  The offical repo is going through changes which searches to rectify that.   Once that refactoring is done,  it can be ported to this repo.   
Otherwise you can always try different distances between tag an antenna, to see if it come alive

hypnoticpattern, 2020-02-22:
I see. Thanks for the quick response!

hypnoticpattern, 2020-05-29:
@iceman1001 I am going to reopen this to get notified if anything changes on the iClass issue.

iceman1001, 2020-05-29:
If you are not working on it,  then noone is,   so I will close this again.

closed 2020-05-29 by iceman1001

hypnoticpattern, 2020-05-30:
My bad. I thought it was on the roadmap

iceman1001, 2020-05-30:
on the internal list, there is 100+ items,  on the public road map about 22,  on the issues list another 32.   And this is voluntary open source project where contributors put their own free time without any payment.  The realistic aspect is that if you want to have some functionality, you better start doing it,  since none of the active contributors has much time for things as it is.   That is the sad truth. 

Finally, we should really remove "general wishlist" with out any spec or willingness from the suggestor from these lists.  We have already started to close issues of that nature.