RfidResearchGroup/proxmark3 #576: Client stuck on Debian 10

CthUlhUzzz, 2020-02-26:
I cloned the latest b87b031 commit and compile it on Debian 10. Client stuck on load (even in standalone mode) and one of the threads load on 100%. But on Kali linux all works fine
[image]

iceman1001, 2020-02-26:
Debian 10 seems to be a troublesum OS...
I had to run the client as _sudo_ even to get it started.

There is isssus with **hf mf autopwn**  and **hf mf hardnested** on Debian 10.

Standalone mode doesnt need a client...
Make sure you remove modem manager and logout / login afterwards.

CthUlhUzzz, 2020-02-26:
The client started with sudo. Any thoughts on what could be the problem?

iceman1001, 2020-02-26:
I have no idea. You haven't followed the instructions for posting new issues, so there is not enough information to go with.

CthUlhUzzz, 2020-02-26:
With root, the client does not load qt5/plugins/platformthemes/KDEPlasmaPlatformTheme.so and the huge list of libraries that depend on it.
`make SKIPQT=1` solve my problem

closed 2020-02-26 by CthUlhUzzz

iceman1001, 2020-02-27:
QT should work on a full fledge *nix system like Debian 10.   
I got it to work by using a different theme,  and running as sudo.