RfidResearchGroup/proxmark3 #577: Keri ID tags not writing properly

swg0101, 2020-02-27:
On the latest master 52452c8ddeb0afe94a43e3ef325436833553710b (and even previous commits), writing a Keri tag to a T55xx based on an internal ID does not result back in an identical lf search read:

```
[usb] pm3 --> lf t5 wip

[=] Begin wiping T55x7 tag

[=] Default configation block 000880E0
[=] Writing page 0  block: 00  data: 0x000880E0
[=] Writing page 0  block: 01  data: 0x00000000
[=] Writing page 0  block: 02  data: 0x00000000
[=] Writing page 0  block: 03  data: 0x00000000
[=] Writing page 0  block: 04  data: 0x00000000
[=] Writing page 0  block: 05  data: 0x00000000
[=] Writing page 0  block: 06  data: 0x00000000
[=] Writing page 0  block: 07  data: 0x00000000
[usb] pm3 --> lf t5 det
    Chip Type      : T55x7
    Modulation     : ASK
    Bit Rate       : 2 - RF/32
    Inverted       : No
    Offset         : 32
    Seq. Term.     : Yes
    Block0         : 0x000880E0
    Downlink Mode  : default/fixed bit length
    Password Set   : No

[usb] pm3 --> lf keri clone 294071780
[=] Preparing to clone KERI to T55x7 with Internal Id: 91872de4
[+] Blk | Data
[+] ----+------------
[+]  00 | 603E1040
[+]  01 | 00000004
[+]  02 | 8C396F27
[+] Success writing to tag
[usb] pm3 --> lf sea
[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...
[=]
[+] KERI Tag Found -- Internal ID: 305194428
[+] Raw: E000000091872DE4
[+] Partial Keri MS decode
[+] BitState ID : -----------0???---01010000101000
[+] BitState FC : -----------------------------?0-
[+] Descrambled MS : FC 4 - Card ID 922664

[+] Valid KERI ID found!
```

Any ideas why this may happen?

swg0101, 2020-02-27:
```
[usb] pm3 --> lf t5 det
    Chip Type      : T55x7
    Modulation     : PSK1
    Bit Rate       : 15 - RF/32
    Inverted       : No
    Offset         : 59
    Seq. Term.     : No
    Block0         : 0x603E1040
    Downlink Mode  : default/fixed bit length
    Password Set   : No

[usb] pm3 --> lf t5 dump
[+] Reading Page 0:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 301F0820 | 00110000000111110000100000100000 | 0..
[+]  01 | 00000002 | 00000000000000000000000000000010 | ....
[+]  02 | C61CB793 | 11000110000111001011011110010011 | ▒.▒▒
[+]  03 | 00000000 | 00000000000000000000000000000000 | ....
[+]  04 | 00000000 | 00000000000000000000000000000000 | ....
[+]  05 | 00000000 | 00000000000000000000000000000000 | ....
[+]  06 | 00000000 | 00000000000000000000000000000000 | ....
[+]  07 | 00000000 | 00000000000000000000000000000000 | ....
[+] Reading Page 1:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 301F0820 | 00110000000111110000100000100000 | 0..
[+]  01 | 701C8068 | 01110000000111001000000001101000 | p.▒h
[+]  02 | 1482941E | 00010100100000101001010000011110 | .▒▒.
[+]  03 | 80500001 | 10000000010100000000000000000001 | ▒P..
[+] saved to json file lf-t55xx-00000002-C61CB793-data.json
[+] saved 12  blocks to text file lf-t55xx-00000002-C61CB793-data.eml
[+] saved 48  bytes to binary file lf-t55xx-00000002-C61CB793-data.bin

```

mwalker33, 2020-02-27:
From your post.
```
Clone cmd   : 294071780 (dec)           10001100001110010110111100100
Internal ID : 91872de4 (hex)         10010001100001110010110111100100
Found       : 305194428	(dec)        10010001100001110010110111100
raw: 1110000000000000000000000000000010010001100001110010110111100100
```
Are you saying the card works ok, just the found ID does not match the one used in the clone command ? (i.e. the encoding is ok, just the decode is off?)

Note: The MS decode can be way off as its not complete and we need some more samples to complete.

swg0101, 2020-02-27:
I actually have not tried the card yet since it's past midnight here. But I thought a read from the original compared to the clone would result in an identical read. I wonder if there was something off in a flag or something that is causing it to write a different value?

mwalker33, 2020-02-27:
(Im not a keri person so just trying to see where is may be out a little)

Are you saying you run lf search and got
[+] KERI Tag Found -- Internal ID: 294071780
so then used 
lf keri clone 294071780
And then got the different result of
[+] KERI Tag Found -- Internal ID: 305194428
Is this your first play with keri, or had it been working in the past ?

swg0101, 2020-02-27:
Yes, the original found the ID that starts with 2, and after cloning to a T55xx and reading it back it turns into an ID that starts with a 3. This is my first Keri IDkey so I haven't had much experience with this.

swg0101, 2020-02-27:
But it looks like the problem is that during the lf keri clone command the left most bit is always set regardless of the provided value. This seems like a logic issue somewhere in the value conversion when the lf keri clone command is used...

swg0101, 2020-02-27:
It looks like that flag is set on line 268 / 269 of cmdlfkeri.c:

```
// MSB is ONE
    internalid |= 0x80000000;
```
Not quite sure why it is needed especially when the original tag has that bit off.

mwalker33, 2020-02-27:
With keri there is different formats including custom (and they can be different lengths).
I think the code to-date is based on Keri-MS (default)
Regardless of if your card has that bit set or not, I would expect that the read and write should match.  (unless there was a reason for it not to ????)
You can see in the code the logic behind the layout.
```
  000000000000000000000000000001XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX111
  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^1###############################^^^
        Preamble block 29 bits of ZEROS
        32 bit Internal ID  (First bit always 1)
        3 bit of 1s in the end
```
which if raw was read as above would show this is correct in the read.

 //get internal id
    uint32_t ID = bytebits_to_byte(DemodBuffer + 29, 32);

But the 3 trailing bits end up being read from the start (maybe due to sync) resulting in the internal ID being the low 32 bits (clearing the msb)

Lets see what the others think.

mwalker33, 2020-02-27:
@iceman
using the low 32 bits from the raw seems to yeild consistent resutls
```
in CmdKeriDemod

    //get internal id
    //uint32_t ID = bytebits_to_byte(DemodBuffer + 29, 32);
    uint32_t ID = raw2;
    ID &= 0x7FFFFFFF;
```

Test run
```
[usb] pm3 --> lf sea
[+] KERI Tag Found -- Internal ID: 1454181112
[+] Raw: E0000000D6AD0AF8

[usb] pm3 --> lf keri clone  1454181112
[=] Preparing to clone KERI to T55x7 with Internal Id: d6ad0af8
[+] Blk | Data
[+] ----+------------
[+]  00 | 603E1040
[+]  01 | 00000006
[+]  02 | B56857C7

[usb] pm3 --> lf sea
[+] KERI Tag Found -- Internal ID: 1454181112
[+] Raw: E0000000D6AD0AF8
```

Thoughts....

iceman1001, 2020-02-27:
Yup,  input should definitly match output for the "simple format",   lowest 32bits.

The more complex format, the Keri MS format is WIP as you very well know  ;)

swg0101, 2020-02-28:
> ```
> in CmdKeriDemod
> 
>     //get internal id
>     //uint32_t ID = bytebits_to_byte(DemodBuffer + 29, 32);
>     uint32_t ID = raw2;
>     ID &= 0x7FFFFFFF;
> ```

This worked for me as well to get consistent results between read/writes.

mwalker33, 2020-02-28:
I can submit a pull request when I get home tonight.
What would be good is if (with the change) you could test if the clone (from the read) works on the reader.  this should confirm if we are correctly reading and writing the correct data.

Thanks.

swg0101, 2020-02-28:
I am currently out of town but should be back sometime next week and should be able to try then. Thanks.

mwalker33, 2020-02-28:
@swg001, the patch has been added.  Lets us know how it goes.

swg0101, 2020-02-28:
Sounds good, I will update as soon as I test this.

swg0101, 2020-03-04:
Tested today - works properly on new commit.