RfidResearchGroup/proxmark3 #582: T5577 password read issues

swg0101, 2020-02-29:
On the current master a46227d96e1a5790fda1a8585be97dac111c3594, it seems like there is an issue where passworded reads from a T55xx does not match its actual contents.

Here's the contents on a non-password read:
```
[usb] pm3 --> lf t5 det
    Chip Type      : T55x7
    Modulation     : PSK1
    Bit Rate       : 15 - RF/32
    Inverted       : No
    Offset         : 28
    Seq. Term.     : No
    Block0         : **0x603E1040**
    Downlink Mode  : default/fixed bit length
    Password Set   : No

[usb] pm3 --> lf t5 read b 0
[+] Reading Page 0:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | **603E1040** | 01100000001111100001000001000000 | `>.@
```

We can see that the block 0 contents match. After protecting the tag, the block 0 contents no longer matches, although the detect shows the block 0 correctly.

```
[usb] pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t5 det p deadbeef
    Chip Type      : T55x7
    Modulation     : PSK1
    Bit Rate       : 15 - RF/32
    Inverted       : No
    Offset         : 59
    Seq. Term.     : No
    Block0         : **0x603E1050**
    Downlink Mode  : default/fixed bit length
    Password Set   : Yes
    Password       : DEADBEEF

[usb] pm3 --> lf t5 read b 0 p deadbeef
[+] Reading Page 0:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[!] Safety check: Could not detect if PWD bit is set in config block. Exits.
[usb] pm3 --> lf t5 read b 0 p deadbeef o
[+] Reading Page 0:
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[=] Safety check overridden - proceeding despite risk
[+]  00 | **301F0828** | 00110000000111110000100000101000 | 0..(
```

mwalker33, 2020-02-29:
This has been discussed many times.
Key challenge is that some modulations are harder to get a good fix/lock on the start of the block data.  when the data has known bit sync patterns (e.g. keri 111 etc) then that can be used to lock the start of the packet (so decoding card types works well).  Also the T55xx config block (page 0 block 0) has a known structure that is used to align that.  The data within all the data blocks is unknown, thus a best effort is made to align based on the "offset" found in the detect of the config. 
I would be very confident that the data on the card will be as written, just the decode in some modulations/speeds can be a little off.  

e.g.
```
Real Config Block Data :  01100000001111100001000001010000
Block  Read Data       : 00110000000111110000100000101000
```
As you can see it is "correct" just 1 bit out, so the password you supplied must have been the correct password as written to block 7

Many people have looked at it for ways to make it better and some are still looking at it.

When I was playing with this, I would set the config of the card back to ASK-32 and it would read perfect, then I was happy the data was correct and set the modulate back to what it was.
While this could be a work around, I really think it would be a very bad idea to write to a card if not requested by the user.

I am sure one day someone we find a way to improve this, but I expect it will take time and research.

If you have any ideas on how to get correct alignment let us know.

swg0101, 2020-02-29:
What I specifically was curious was how the lf t5 detect get this right but the read or dump commands would always be off. Do you think this would still be modulation related?

mwalker33, 2020-02-29:
I am doing this from memory, so my not be 100%
When you do the lf t55 det
It will read block 0 page 0 then do a "special" check.
It will do some high level checks to rule out (in) some modulation types, then based on whats left, do some speed checks (if it can), then simply type all combos of what could still be valid.
For each modulation type and speed it will then check at each offset if the data fits a valid T55xx config block.  If it is, it adds that to a list of possible configs.
If there is only 1 (which seems to be every time for me), then that is reported back as the config.
At the same time it takes note of the offset (very often 32/33 bits).
When a normal read is made, it will just a read of the block data, demodulate with what the detected with the detect and display the data from bit at offset x as found in the config/detect.  
But lets say the card started modulating a little earlier or later, then the data will be off +/- some bits.
So when you perform a read of block 0, it wont do all the config checks (as that should have been from the config).

swg0101, 2020-02-29:
I thought if a particular valid block 0 and offsets are detected via lf t5 detect, the reads would also use that information to derive valid information from it? Otherwise, there is a high risk of a dump and load not resulting in valid data...

mwalker33, 2020-02-29:
It does used the information from the detect. 
When you read from the t55xx it will send the requested block (32 bits) in a loop until the field drops.
So an offset of 32 bits means demodulate the bit stream.  then go to bit x and take the next 32 bits as the data for the block.  but if for some reason the demodulate missed the first bit, or there was something else that got read as an extra bit at the start, (or more), then the offset could be a little out on each read (remember the raw modulated data will be the digital waveform, so a bit will span over many samples, the slower the speed the more samples per bit.  
What you sometimes see is the actual data can affect the demodulate start location.  
e.g. Lets say we have the value 00001111 so the first 16 bits are zero.
Lets say that a 0 is transmitted as no modulation and a 1 is high.  Then the first 16 bits will just be a flat line.  now, if it was late "transmitting" then you could have another bit worth of nothing at the start of the samples.  How would you know were the first bit started.
In a perfect world, we would know that the first bit sample will start at the exact same spot every time.  But, if you look at the sniff/snoop wave form, you will see the start can be a little hard to read (I believe) this is due to things settling.   e.g. i expect the card will put more load on the reader while it charges up, then it could drop while using its power to modulate. 

This is a little hard to explain until you look at the raw samples.

iceman1001, 2020-02-29:
Since there are no real new password read issues, just old known problematics, I suggest we close this one.

closed 2020-03-04 by swg0101