RfidResearchGroup/proxmark3 #583: hf mfdes auth 2 1|2|3 0   correct DES key   returns SESSION :00 00 00 00 00 00 00 00

michal25, 2020-02-29:
This
[usb] pm3 --> hf mfdes auth 2 1 0 eeeeeeeeeeeeeeee
DES selected          
  Key        :EE EE EE EE EE EE EE EE           
  SESSION    :00 00 00 00 00 00 00 00           
-------------------------------------------------------------          
-------------------------------------------------------------          
[usb] pm3 --> hf mfdes enum
          
-- Desfire Enumerate Applications ---------------------------          
-------------------------------------------------------------          
 Aid 0 : 56 34 12           
 AMK - Application Master Key settings          
Changekey Access rights          
-- Authentication with the specified key is necessary to change any key. A change key and a PICC master key (CMK) can only be changed after authentication with the master key. For keys other then the master or change key, an authentication with the same key is needed.          
          
   0x08 Configuration changeable       : NO          
   0x04 AMK required for create/delete : NO          
   0x02 Directory list access with AMK : NO          
   0x01 AMK is changeable              : NO          
          
     Max number of keys  : 164          
     Application Master key Version  : 160 (0xa0)          
-------------------------------------------------------------          
#db# Can't select card          
[!]    Can't get file ids           
[!]    Can't get ISO file ids           
------------------------------------------------------------- 

But, the key is correct. Maybe some bug in the Pm3 code? Or, my misunderstanding?

OS
Ubuntu 18.04 LTS 64 bit desktop

[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]           

 [ CLIENT ]           
  client: RRG/Iceman          
  compiled with GCC 7.4.0 OS:Linux ARCH:x86_64          

 [ PROXMARK3 RDV4 ]           
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK3 RDV4 Extras ]           
  FPC USART for BT add-on support: present           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/52452c8d 2020-02-27 12:23:35
       os: RRG/Iceman/master/52452c8d 2020-02-27 12:23:45
  compiled with GCC 6.3.1 20170620

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-02-22 at 12:51:14
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 286320 bytes (55%) Free: 237968 bytes (45%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory 

[usb] pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing           
#db#   tracing ................0          
#db#   traceLen ...............487          
#db# Currently loaded FPGA image           
#db#   mode.................... HF image built for 2s30vq100 on 2020-01-12 at 15:31:16          
#db# Flash memory           
#db#   Baudrate................24 MHz           
#db#   Init....................OK           
#db#   Memory size.............2 mbits / 256 kb           
#db#   Unique ID...............0xD5690C23DF31492A          
#db# Smart card module (ISO 7816)           
#db#   version.................v3.10           
#db# LF Sampling config           
#db#   [q] divisor.............95 ( 125.00 kHz )          
#db#   [b] bits per sample.....8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........No          
#db#   [t] trigger threshold...0          
#db#   [s] samples to skip.....0           
#db# LF T55XX config           
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]          
#db#            mode            |start|write|write|write| read|write|write          
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3          
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------          
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |           
#db#     long leading reference |  29 |  17 |  15 |  47 |  15 | N/A | N/A |           
#db#               leading zero |  29 |  17 |  15 |  40 |  15 | N/A | N/A |           
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |           
#db#           
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......321536          
#db#   Transfer Speed PM3 -> Client = 643072 bytes/s          
#db# Various           
#db#   DBGLEVEL................1          
#db#   ToSendMax...............39          
#db#   ToSendBit...............8          
#db#   ToSend BUFFERSIZE.......2308          
#db#   Slow clock..............30174 Hz          
#db# Installed StandAlone Mode           
#db#   HF Mifare sniff/simulation - (Craig Young)          
#db# Flash memory dictionary loaded           
#db#   Mifare..................906 keys          
#db#   T55x7...................109 keys          
#db#   iClass..................7 keys

michal25, 2020-02-29:
And info about smartcard
<pre>
[usb] pm3 --> hf mfdes info
          
-- Desfire Information --------------------------------------          
-------------------------------------------------------------          
  UID                : 04 11 45 92 9F 65 80           
  Batch number       : CE EC D3 5D 60           
  Production date    : week 21, 2019          
  -----------------------------------------------------------          
  Hardware Information          
      Vendor Id      : NXP Semiconductors Germany          
      Type           : 0x01          
      Subtype        : 0x01          
      Version        : 18.0 (Desfire EV2)          
      Storage size   : 0x18 (4096 bytes)          
      Protocol       : 0x05 (ISO 14443-3, 14443-4)          
  -----------------------------------------------------------          
  Software Information          
      Vendor Id      : NXP Semiconductors Germany          
      Type           : 0x01          
      Subtype        : 0x01          
      Version        : 2.1          
      storage size   : 0x18 (4096 bytes)          
      Protocol       : 0x05 (ISO 14443-3, 14443-4)          
-------------------------------------------------------------          
 CMK - PICC, Card Master Key settings           
          
   [0x08] Configuration changeable       : YES          
   [0x04] CMK required for create/delete : NO          
   [0x02] Directory list access with CMK : NO          
   [0x01] CMK is changeable              : YES          
          
   Max number of keys       : 174          
   Master key Version       : 0 (0x00)          
   ----------------------------------------------------------          
   [0x0A] Authenticate      : YES          
   [0x1A] Authenticate ISO  : YES          
   [0xAA] Authenticate AES  : NO          
          
   ----------------------------------------------------------          
   Available free memory on card       : 4896 bytes          
-------------------------------------------------------------
</pre>

iceman1001, 2020-02-29:
The des commands isn't implemented much.  At this moment I think only DES option is implemented in ```hf mfdes auth```

closed 2020-02-29 by iceman1001