RfidResearchGroup/proxmark3 #584: USB description error

Windslab, 2020-03-02:
After updating to current firmware got an USB description error. Have backed again to older version and have positive dynamics.

iceman1001, 2020-03-02:
We would need more details than that in order to look into this claim.

Windslab, 2020-03-02:
Proxmark easy jumping with USB Descriptor after flashing.
*Sorry NO Description is DESCRIPTOR
What is the orfografii correction doing is not good sometimes

iceman1001, 2020-03-02:
You would need to write which steps you did and your output from the commands you ran.

None of my proxmark3 get a bad usb descriptor when flashing.  It could be related to Pm3 Easy clone hardware.  In such case nothing to do here.

Windslab, 2020-03-02:
My steps is: Compile> FLASH - All.bat and obtain err of USB Descriptor
After about 5 time poweroff I see COM in my hardware list.
Have you did any changers with communication setting last time? Before thise relise all been good.

You can't say "Pm3 Easy clone hardware" it is an open source project with shared PCBs. Anybody can reproduce it.

iceman1001, 2020-03-02:
Flash All.bat?  Thats not a file in this repo.  You seem to be using some precompiled binaries.

Nothing changed in communication setting last time. 

The phrase "pm3 easy clone hardware" come from hardware manufacturers who produce crappy products.  Nothing about being open source project at all.

closed 2020-03-02 by iceman1001