RfidResearchGroup/proxmark3 #587: [Idea] Choose between Proxmarks

joanbono, 2020-03-03:
Hi guys,

I was thinking about adding some optional apartment (maybe under Linux Installation instructions) about adding some alias to the proxmarks *Useful if you have more than 1).

So, basically, the idea is to create a rule on `/etc/udev/rules.d/`:

Eg. My `/etc/udev/rules.d/99-proxmark3.rules`:

``` bash 
#Proxmark3 Rvd2 Elechouse
... 

# Proxmark3 Easy
SUBSYSTEM=="tty", ATTRS{idVendor}=="9ac4", ATTRS{idProduct}=="4b8f", GROUP="dialout", MODE="0666", SYMLINK+="pm3easy"

# Proxmark3 Rvd4
...
``` 

After this, the access to the proxmark will be easier to identify to which one are you connecting:

```
./proxmark3 /dev/pm3easy
```  

What do you think? Should I go for it or it's not necessary at all?

Regards

iceman1001, 2020-03-03:
the only one you could differentiate would be  pm3easy with strange fw vs offical/iceman fw.
Once you flash open source fw onto the device,  you will not identify it as pm3easy any longer.

So I don't see the value for it.  I understand where you coming from with wanting to identify your device.  The "symlink alias" will give you pm3-n  as names.   We just don't identify on vid/pid in the client.  That is for the OS.

There is a slight change that if you give the dev/tty name _pm3easy_  users might think that "aha" the os knows my device,  which will be confusing when the OS starts giving them  _pm3-0_ instead...

joanbono, 2020-03-03:
Okay, gotcha :) 

Closing this. Thanks for the answer!

closed 2020-03-03 by joanbono