RfidResearchGroup/proxmark3 #588: The "lf search" plus version(Automated iteration division value)

xianglin1998, 2020-03-04:
The frequency of some ID cards is relatively rare. At this time, the standard 125kHz search will not work. I improved this situation and confirmed the command as "LF search + +"

This is lf search++ of RFID Tools client. Because I changed the communication part of the library a lot, I couldn't submit it directly.

The CommandTable:
`{"search++",  CmdLFSearchPlus,  IfPm3Lf,         "You can automatically iterate the divisor and execute the LF search command, and the iteration will end after the card is successfully searched."},`

Impl1:
`void recovery_divisor() {
    sample_config config = {
            .decimation = -1,
            .bits_per_sample = -1,
            .averaging = -1,
            .divisor = 95,
            .trigger_threshold = -1,
            .samples_to_skip = -1,
            .verbose = false
    };
    lf_config(&config);
}`

Impl2:
`// DXL, The lf search plus version!
int CmdLFSearchPlus(const char *Cmd) {
    int retval = PM3_SUCCESS;
    // we have some default divisor values
    // divisor : frequency(khz)
    // 95 -> 125.00
    // 88 -> 134.83
    // 47 -> 250.00
    // 31 -> 375.00
    // 23 -> 500.00
    int default_divisor[] = {95, 88, 47, 31, 23};
    // build a config struct
    sample_config config = {
            .decimation = -1,
            .bits_per_sample = -1,
            .averaging = -1,
            .divisor = -1,
            .trigger_threshold = -1,
            .samples_to_skip = -1,
            .verbose = true
    };
    // Iteration defaults
    for (int i = 0; i < sizeof(default_divisor) / sizeof(int); ++i) {
        if (kbd_enter_pressed()) {
            PrintAndLogEx(NORMAL, "%s", "Keyboard pressed Done.");
            recovery_divisor();
            return retval;
        }
        // Try to change config!
        config.divisor = (int16_t) default_divisor[i];
        if (lf_config(&config) == PM3_SUCCESS) {
            // The config for pm3 is changed, we can trying search!
            retval = CmdLFfind(Cmd);
            if (retval == PM3_SUCCESS) {
                recovery_divisor();
                return retval;
            } else {
                PrintAndLogEx(NORMAL, "\n");
            }
        } else {
            recovery_divisor();
            return PM3_ESOFT;
        }
    }
    // Iterate all values(from 19 to 255)
    for (int i = 19; i <= 255; ++i) {
        if (kbd_enter_pressed()) {
            PrintAndLogEx(NORMAL, "%s", "Keyboard pressed Done.");
            recovery_divisor();
            return retval;
        }
        for (int j = 0; j < sizeof(default_divisor) / sizeof(int); ++j) {
            if (i == default_divisor[i]) {
                continue;
            }
        }
        config.divisor = (int16_t) i;
        if (lf_config(&config) == PM3_SUCCESS) {
            retval = CmdLFfind(Cmd);
            if (retval == PM3_SUCCESS) {
                break;
            }
        } else {
            recovery_divisor();
            return PM3_ESOFT;
        }
    }
    recovery_divisor();
    return retval;
}`

iceman1001, 2020-03-04:
Cool idea,   let me just ask about how much the changing of frequency during **lf search** helps?

Have you tried your search++ on cards that has 250 kHz, 375 kHz, 500 kHz?
```
   // 95 -> 125.00
    // 88 -> 134.83
    // 47 -> 250.00
    // 31 -> 375.00
    // 23 -> 500.00
```
If you did,  did those cards hadve a modulation that was identified with the currect implemented demods in the pm3 client?

Since basically you are running **lf search** with different frequence and run all known demodulations against it over and over.

I am curious to hear if you have cards with those frequences and which results you got from your extended search.

xianglin1998, 2020-03-05:
Yes, I have tried to search tag with 134khz and 250kHz frequencies. When I configure it as 88, I can search tag with 134khz frequencies, and 250kHz frequencies.

Similarly, 250kHz tag need to be configured with 47 divisors to search for tag.

The modulation of these special frequency ID cards is no different from ordinary ID cards, that is, the capacitive inductance on the hardware has been changed, resulting in that the RF frequency cannot be correctly matched by the client of PM3 (default 125khz). We need to perform frequency matching until The card is correctly communicated.

We have tested that it is indeed available, for cards with special frequencies, when the standard search instructions are not available.

iceman1001, 2020-03-05:
I am very well aware of the purpose of the divisor. That wasn't my question.

My question is if you actually tested against a 250khz card and got it to be identified with the known demodulations inside **lf search**.   

The problem with your experiments as I seem it,  is that you change divisor but your hardware antenna is still tuned for 125 kHz.   

Let me demonstrate it with a picture.
[image]

What kind of voltage are you getting out of your antenna at those frequencies to power a tag and is the build in demodulation able to do its work with such time voltage changes?

xianglin1998, 2020-03-05:
Today, we tested the special frequency tag from 125khz to 500khz, but the result is not satisfactory. We need to replace the antenna, hahaha, so this instruction can be kept temporarily, and we hope there will be a hardware solution for automatic frequency tuning in the future.

xianglin1998, 2020-03-05:
;) Thank you for checking.

closed 2020-03-06 by xianglin1998

iceman1001, 2020-03-06:
Special frequency tag,   interesting,  link to a page?  datasheet?

xianglin1998, 2020-03-06:
Yes is very interesting.

You can buy it on taobao:

https://detail.tmall.com/item.htm?spm=a230r.1.14.36.78b361c0gutHid&id=610148446963&ns=1&abbucket=1&skuId=4460492583507

xianglin1998, 2020-03-06:
But I don't have a datasheet.

iceman1001, 2020-03-06:
Cool,  I am curious on what type of keyfob.  Like what it can do.

xianglin1998, 2020-03-06:
They look like ordinary tags, but in fact, the coil and capacitance parameters have been adjusted. It needs a specific card reader to read the card. On the ordinary 125 kHz card reader, it does not work.

iceman1001, 2020-03-06:
yes,  but are they EM4100 / t5577 / prox ?  What can the keyfob do besides 250/375/500?

xianglin1998, 2020-03-06:
It is can change type at em & t55xx between，like your t5577, the type of card has nothing to do with the frequency.

iceman1001, 2020-03-06:
cool,  now I will need to get one of those keyfobs...  and an tunable antenna.

xianglin1998, 2020-03-06:
I'll ask Olaf to make antennas for these frequencies, and then I'll pack the antennas and keyfob and send them to you.

:)

xianglin1998, 2020-03-06:
some 250, 375, 500khz keyfobs.

xianglin1998, 2020-03-06:
and 250, 375, 500khz antennas .

iceman1001, 2020-03-06:
Sounds like a great plan.
Push his limits by saying one antenna with auto-tuning and dynamic frequency selection. 
That will be a challenge for him,  he likes that.