RfidResearchGroup/proxmark3 #589: About Mifare Desfire auth PM3 implementation (desfire commands)

michal25, 2020-03-05:
Hello. I'm interested in Mifare Desfire auth PM3 implementation. 
At this momen we can use the client commands
hf mfdes auth

Usage:  hf mfdes auth <1|2|3> <1|2|3|4> <keyno> <key>           
            Auth modes          
                 1 = normal, 2 = iso, 3 = aes          
            Crypto          
                 1 = DES 2 = 3DES 3 = 3K3DES 4 = AES

But the code is not fully implemented and you can obtain this result
[usb] pm3 --> hf mfdes auth 2 2 0 ffffffffffffffffffffffffffffffff
3DES selected          
  Key        :FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF           
  SESSION    :00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
-------------------------------------------------------------          
-------------------------------------------------------------
when the keyno and key are correct.

I collected some documentation about the des, 3des and AES auth process, they are located here
http://www.proxmark.org/files/Documents/13.56%20MHz%20-%20MIFARE%20DESFire/

What is needed:
1. More documentation
2. Researchers with testing polygon (2xPM3 + Atmel studio, Chameleon, RFID readers, testing Mifare Desfire, Mifare Desfire EV1,EV2,(EV3?) and Mifare Desfire Light smartcards.
3. Researchers with good c-code knowledge.
4. Every help will be welcome

bkerler, 2020-04-07:
We are already working on fixing the Mifare Desfire auth. The aes auth has already been fixed and the des/tfdes/3des implementation is WIP.

michal25, 2020-04-08:
Great!!
As I can see, the PM3 code reports now some small errors on **hf mfdes info**  and **hf mfdes enum**  on desfire ev1  and desfire ev2 cards.

[usb] pm3 --> hf mfdes info
          
[=] --- Tag Information ---------------------------          
[=] -------------------------------------------------------------          
[+]               UID: 04 11 45 92 9F 65 80           
[+]      Batch number: CE EC D3 5D 60           
[+]   Production date: week 21 / 2019           
          
[=] --- Hardware Information           
[=]      Vendor Id: NXP Semiconductors Germany           
[=]           Type: 0x01           
[=]        Subtype: 0x01           
[=]        Version: 12.0 ( DESFire EV2 )          
[=]   Storage size: 0x18 ( 4096 bytes )          
[=]       Protocol: 0x05 ( ISO 14443-3, 14443-4 )          
          
[=] --- Software Information           
[=]      Vendor Id: NXP Semiconductors Germany           
[=]           Type: 0x01           
[=]        Subtype: 0x01           
[=]        Version: 2.1           
[=]   Storage size: 0x18 ( 4096 bytes )          
[=]       Protocol: 0x05 ( ISO 14443-3, 14443-4 )          
          
[=] --- Card capabilities           
          
[=] --- Tag Signature           
[=]  IC signature public key name: DESFire EV2          
[=] IC signature public key value: 04B304DC4C615F5326FE9383DDEC9AA8          
[=]                              : 26FE9383DDEC9AA892DF3A57FA7FFB32          
[=]                              : 92DF3A57FA7FFB3276192BC0EAA252ED          
[=]                              : 76192BC0EAA252ED45A865E3B093A3D0          
[=]     Elliptic curve parameters: NID_secp224r1          
[=]              TAG IC Signature: EC FF 55 85 8B C8 13 BB B8 46 41 F9 33 19 52 F6          
[=]                              : 9A F0 F5 5F 03 AB 69 B4 D4 EB 8A 75 2E 98 1A D7          
[=]                              : 56 B5 DE EE 05 79 60 EC 92 F3 A7 C8 E4 A9 A3 D2          
[=]                              : 84 D3 3C BB 12 9C 0A 04          
[+]         Signature verified: successful           
[+]    Number of Masterkeys                  : 1           
[+]    [0x08] Configuration changeable       : YES           
[+]    [0x04] CMK required for create/delete : YES           
[+]    [0x02] Directory list access with CMK : YES           
[+]    [0x01] CMK is changeable              : YES           
[+]    Operation of PICC master key          : (3)DES           
[+]    PICC Master key Version               : 0 (0x00)           
[=]    ----------------------------------------------------------          
[+]    [0x0A] Authenticate      : YES           
[+]    [0x1A] Authenticate ISO  : NO          
[+]    [0xAA] Authenticate AES  : NO          
[=] -------------------------------------------------------------          
          
[=] --- Free memory           
[+]    Available free memory on card         : 4896 bytes           
[=] -------------------------------------------------------------          
[usb] pm3 --> hf mfdes enum
          
[=] -- Mifare DESFire Enumerate applications --------------------          
[=] -------------------------------------------------------------          
[+]  Tag report 1 application           
          
[+] --- AMK - Application Master Key settings           
[+]   AID : 56 34 12           
[!]  ⚠️  [get_desfire_keysettings] Authentication error           
[!]  ⚠️     Can't read Application Master key settings           
[=] -------------------------------------------------------------          
[=]   Application keys          
[+]    Key [0]  Version : 160 (0xa0)          
[=] -------------------------------------------------------------

michal25, 2020-04-08:
And for EV1
[!]  ⚠️  Command unsuccessful           
[usb] pm3 --> hf mfdes enum
          
[=] -- Mifare DESFire Enumerate applications --------------------          
[=] -------------------------------------------------------------          
[+]  Tag report 2 applications          
          
[+] --- AMK - Application Master Key settings           
[+]   AID : 50 11 F2           
[+]   AID Key settings           : 0b          
[+]   Max number of keys in AID  : 5          
[=] -------------------------------------------------------------          
[+]   Changekey Access rights          
[+]   -- AMK authentication is necessary to change any key (default)          
[+]    [0x08] Configuration changeable       : YES           
[+]    [0x04] AMK required for create/delete : YES          
[+]    [0x02] Directory list access with AMK : NO          
[+]    [0x01] AMK is changeable              : YES           
[=] -------------------------------------------------------------          
[=]   Application keys          
[+]    Key [0]  Version : 0 (0x00)          
[+]    Key [1]  Version : 0 (0x00)          
[+]    Key [2]  Version : 0 (0x00)          
[+]    Key [3]  Version : 0 (0x00)          
[+]    Key [4]  Version : 0 (0x00)          
[+]  Tag report 1 file           
[+]    Fileid 0 (0x00)          
[=]   Settings [7] 01 03 10 20 E0 01 00          
          
[+] --- AMK - Application Master Key settings           
[+]   AID : FF FF FF           
[+]   AID Key settings           : 0b          
[+]   Max number of keys in AID  : 5          
[=] -------------------------------------------------------------          
[+]   Changekey Access rights          
[+]   -- AMK authentication is necessary to change any key (default)          
[+]    [0x08] Configuration changeable       : YES           
[+]    [0x04] AMK required for create/delete : YES          
[+]    [0x02] Directory list access with AMK : NO          
[+]    [0x01] AMK is changeable              : YES           
[=] -------------------------------------------------------------          
[=]   Application keys          
[+]    Key [0]  Version : 1 (0x01)          
[+]    Key [1]  Version : 0 (0x00)          
[+]    Key [2]  Version : 1 (0x01)          
[+]    Key [3]  Version : 0 (0x00)          
[+]    Key [4]  Version : 0 (0x00)          
[+]  Tag report 3 files          
[+]    Fileid 0 (0x00)          
[=]   Settings [17] 02 03 00 40 03 00 00 00 0A 00 00 00 00 00 00 00 00          
[+]    Fileid 1 (0x01)          
[=]   Settings [7] 00 03 00 32 30 00 00          
[+]    Fileid 2 (0x02)          
[=]   Settings [7] 00 03 00 12 30 00 00          
[=] -------------------------------------------------------------

michal25, 2020-04-08:
hf mfdes info
hf mfdes enum
seems to be functional for ev1

good job

I found a little error on
hf mfdes enum 
for desfire ev2

[usb] pm3 --> hf mfdes enum
          
[=] -- Mifare DESFire Enumerate applications --------------------          
[=] -------------------------------------------------------------          
[+]  Tag report 1 application           
          
[+] --- AMK - Application Master Key settings           
[+]   AID : 56 34 12           
[!]  ⚠️  [get_desfire_keysettings] Authentication error           
[!]  ⚠️     Can't read Application Master key settings           
[=] -------------------------------------------------------------          
[=]   Application keys          
[+]    Key [0]  Version : 160 (0xa0)          
[=] -------------------------------------------------------------

and
[usb] pm3 --> hf mfdes enum
#db# Can't select card          
*** buffer overflow detected ***: /usr/local/bin/proxmark3 terminated
/usr/local/bin/pm3: řádek 66:   583 Neúspěšně ukončen (SIGABRT)        (core dumped [obraz paměti uložen]) $CLIENT "$@"

bkerler, 2020-04-09:
I'm currently rewrite huge parts of the code. The upcoming versions should be much better.

bkerler, 2020-04-11:
Please test the latest commit as many things have been fixed and improved :)

michal25, 2020-04-15:
Implementation for desfire ev1 looks good and commands
hf mfdes info
hf mfdes enum

returns correct data

desfire ev2 reports this problem on 
hf mfdes enum

[usb] pm3 --> hf mfdes enum
          
[=] -- Mifare DESFire Enumerate applications --------------------          
[=] -------------------------------------------------------------          
[+]  Tag report 1 application           
          
[+] --- AMK - Application Master Key settings           
[+]   AID : 123456           
[+]   AID Function Cluster 0x12: reserved           
[!]  ⚠️     Can't read Application Master key settings           
[!!]  🚨
[!]  ⚠️     Can't read AID master key version. Trying all keys          
[!]  ⚠️     Can't get file ids -> Current authentication status does not allow the requested command 

hf mfdes info looks good for ev2
[usb] pm3 --> hf mfdes info
          
[=] --- Tag Information ---------------------------          
[=] -------------------------------------------------------------          
[+]               UID: 04 11 45 92 9F 65 80            
[+]      Batch number: CE EC D3 5D 60            
[+]   Production date: week 21 / 2019           
          
[=] --- Hardware Information           
[=]      Vendor Id: NXP Semiconductors Germany           
[=]           Type: 0x01           
[=]        Subtype: 0x01           
[=]        Version: 12.0 ( DESFire EV2 )          
[=]   Storage size: 0x18 ( 4096 bytes )          
[=]       Protocol: 0x05 ( ISO 14443-2, 14443-3 )          
          
[=] --- Software Information           
[=]      Vendor Id: NXP Semiconductors Germany           
[=]           Type: 0x01           
[=]        Subtype: 0x01           
[=]        Version: 2.1           
[=]   Storage size: 0x18 ( 4096 bytes )          
[=]       Protocol: 0x05 ( ISO 14443-3, 14443-4 )          
          
[=] --- Card capabilities           
          
[=] --- Tag Signature           
[=]  IC signature public key name: DESFire EV2           
[=] IC signature public key value: 04B304DC4C615F5326FE9383DDEC9AA8          
[=]                              : 26FE9383DDEC9AA892DF3A57FA7FFB32          
[=]                              : 92DF3A57FA7FFB3276192BC0EAA252ED          
[=]                              : 76192BC0EAA252ED45A865E3B093A3D0          
[=]     Elliptic curve parameters: NID_secp224r1          
[=]              TAG IC Signature: ECFF55858BC813BBB84641F9331952F6          
[=]                              : 9AF0F55F03AB69B4D4EB8A752E981AD7          
[=]                              : 56B5DEEE057960EC92F3A7C8E4A9A3D2          
[=]                              : 84D33CBB129C0A04          
[+]            Signature verified: successful           
[+]    Number of Masterkeys                  : 1           
[+]    [0x08] Configuration changeable       : YES           
[+]    [0x04] CMK required for create/delete : YES           
[+]    [0x02] Directory list access with CMK : YES           
[+]    [0x01] CMK is changeable              : YES           
[+]    Operation of PICC master key          : (3)DES           
[+]    PICC Master key Version               : 0 (0x00)           
[=]    ----------------------------------------------------------          
[+]    [0x0A] Authenticate      : YES           
[+]    [0x1A] Authenticate ISO  : YES           
[+]    [0xAA] Authenticate AES  : NO          
[=] -------------------------------------------------------------          
          
[=] --- Free memory           
[+]    Available free memory on card         : 4896 bytes 

For the chip DESFire MF3ICD40 hf mfdes enum report troubles
[usb] pm3 --> hf mfdes enum
[!]  ⚠️  Can't get DF Names 

hf mfdes info is ok for DESFire MF3ICD40
[usb] pm3 --> hf mfdes info
          
[=] --- Tag Information ---------------------------          
[=] -------------------------------------------------------------          
[+]               UID: 04 5C 41 4A 96 26 80            
[+]      Batch number: BA 24 13 8F 80            
[+]   Production date: week 04 / 2011           
          
[=] --- Hardware Information           
[=]      Vendor Id: NXP Semiconductors Germany           
[=]           Type: 0x01           
[=]        Subtype: 0x01           
[=]        Version: 0.2 ( DESFire MF3ICD40 )          
[=]   Storage size: 0x18 ( 4096 bytes )          
[=]       Protocol: 0x05 ( ISO 14443-2, 14443-3 )          
          
[=] --- Software Information           
[=]      Vendor Id: NXP Semiconductors Germany           
[=]           Type: 0x01           
[=]        Subtype: 0x01           
[=]        Version: 0.6           
[=]   Storage size: 0x18 ( 4096 bytes )          
[=]       Protocol: 0x05 ( ISO 14443-3, 14443-4 )          
          
[=] --- Card capabilities           
[=]     0.6 - DESFire MF3ICD40, Add ISO/IEC 7816 command set compatibility          
[+]    Number of Masterkeys                  : 1           
[+]    [0x08] Configuration changeable       : YES           
[+]    [0x04] CMK required for create/delete : NO          
[+]    [0x02] Directory list access with CMK : YES           
[+]    [0x01] CMK is changeable              : YES           
[+]    Operation of PICC master key          : (3)DES           
[+]    PICC Master key Version               : 0 (0x00)           
[=]    ----------------------------------------------------------          
[+]    [0x0A] Authenticate      : YES           
[+]    [0x1A] Authenticate ISO  : NO          
[+]    [0xAA] Authenticate AES  : NO          
[=] -------------------------------------------------------------          
          
[=] --- Free memory           
[+]    Card doesn't support 'free mem' cmd          
[=] -------------------------------------------------------------

michal25, 2020-04-19:
some mistake in the pm3 code?

cd /opt/proxmark3
git pull

make clean&&make all
[-] CXX src/proxguiqt.cpp
[-] CXX proxguiqt.moc.cpp
proxguiqt.moc.cpp:9:10: fatal error: proxguiqt.h: Adresář nebo soubor neexistuje
 #include "proxguiqt.h"
          ^~~~~~~~~~~~~
compilation terminated.
Makefile:393: recipe for target 'obj/proxguiqt.moc.o' failed
make[1]: *** [obj/proxguiqt.moc.o] Error 1
Makefile:97: recipe for target 'client/all' failed
make: *** [client/all] Error 2

doegox, 2020-04-19:
the repo compiles fine but it went through some reorganization.
try from a fresh git clone

doegox, 2020-04-19:
my guess is that you've an old ``client/proxguiqt.moc.cpp``. you must erase it.
it should go like this:
```
[-] CXX src/proxguiqt.cpp
[-] MOC src/proxguiqt.moc.cpp
[-] CXX src/proxgui.cpp
[-] CXX src/proxguiqt.moc.cpp
```

doegox, 2020-04-19:
I pushed a change in ``make clean`` to cope with old temp files as well

michal25, 2020-04-27:
For DESFIRE MF3ICD40 chip
hf mfdes info is working but hf mfdes enum not working.

[usb] pm3 --> hf mfdes enum
[!] ⚠️  Can't get DF Names          
[usb] pm3 -->

[usb] pm3 --> hf mfdes info
          
[=] --- Tag Information ---------------------------          
[=] -------------------------------------------------------------          
[+]               UID: 04 56 35 E9 76 1B 80           
[+]      Batch number: 8F D4 15 59 60           
[+]   Production date: week 26 / 2008          
          
[=] --- Hardware Information          
[=]      Vendor Id: NXP Semiconductors Germany          
[=]           Type: 0x01          
[=]        Subtype: 0x01          
[=]        Version: 0.2 (DESFire MF3ICD40)          
[=]   Storage size: 0x18 (4096 bytes)          
[=]       Protocol: 0x05 (ISO 14443-2, 14443-3)          
          
[=] --- Software Information          
[=]      Vendor Id: NXP Semiconductors Germany          
[=]           Type: 0x01          
[=]        Subtype: 0x01          
[=]        Version: 0.6          
[=]   Storage size: 0x18 (4096 bytes)          
[=]       Protocol: 0x05 (ISO 14443-3, 14443-4)          
          
[=] --- Card capabilities          
[=]     0.6 - DESFire MF3ICD40, Add ISO/IEC 7816 command set compatibility          
[+]    Number of Masterkeys                  : 1          
[+]    [0x08] Configuration changeable       : YES          
[+]    [0x04] CMK required for create/delete : NO          
[+]    [0x02] Directory list access with CMK : YES          
[+]    [0x01] CMK is changeable              : YES          
[+]    Operation of PICC master key          : (3)DES          
[+]    PICC Master key Version               : 0 (0x00)          
[=]    ----------------------------------------------------------          
[+]    [0x0A] Authenticate      : YES          
[+]    [0x1A] Authenticate ISO  : NO          
[+]    [0xAA] Authenticate AES  : NO          
[=] -------------------------------------------------------------          
          
[=] --- Free memory          
[+]    Card doesn't support 'free mem' cmd          
[=] -------------------------------------------------------------

iceman1001, 2020-04-27:
yeah, I suspect that the first generation DESFire cards doesn't understand wrapped native apdu's,  but only native apdu's.

doegox, 2020-04-27:
Hmm from 2008 doc, the  optional wrapping of the native DESFire APDU format" was already present in MF3ICDxx

michal25, 2020-04-27:
Well, the PM3 code from February 2020 worked well and I could enumerate the application ID's on the  DESFIRE MF3ICD40 chip.

iceman1001, 2020-04-27:
Feel free to contribute with a fix.

doegox, 2020-04-27:
According to git bisect, bug appears in a3ea353dab97a2ed301783daf61bbd8c1e9129ae from April 8
Poking @bkerler

bkerler, 2020-04-27:
Seems the problem is getting the iso df name. @michal25 can you upload the log running enum with "data setdebugmode 2" being set before ?

doegox, 2020-04-27:
Hmm sorry  a3ea353 is probably not the cause, it's properly aborting the enum command while before the error was silently ignored.

michal25, 2020-04-28:
Here is the log for MF3ICD40 chip
[usb] pm3 --> data setdebugmode 2
[usb] pm3 --> hf mfdes enum
[+] >>>> 90 6A 00 00 00           
[+] <<<< FF FF FF 50 11 F2 91 00           
[+] >>>> 90 6D 00 00 00           
[+] <<<< 91 1C           
[#] Command code not supported          
[!] ⚠️  Can't get DF Names

bkerler, 2020-04-28:
Pull request #721 should fix this issue. It does return as df iso names aren't supported although
they should be supported. It will now ignore that error and set df name count to 0 instead. Please try and report back :)

doegox, 2020-04-28:
>  It does return as df iso names aren't supported although they should be supported

I thought so but http://neteril.org/files/M075031_desfire.pdf p19 
*3.8 MF3 IC D40 Command Set Overview – PICC Level Commands*
doesn't have the 0x6D command, which appears only in later versions

michal25, 2020-04-28:
git pull at 28.04.2020 19.20  Prague time
Still not working

One chip
[usb] pm3 --> hf mfdes enum
[+] >>>> 90 6A 00 00 00           
[+] <<<< FF FF FF 50 11 F2 91 00           
[+] >>>> 90 6D 00 00 00           
[+] <<<< 91 1C           
[#] Command code not supported          
[!] ⚠️  Can't get DF Names

Second chip
[usb] pm3 --> hf mfdes enum
[+] >>>> 90 6A 00 00 00           
[+] <<<< 2F 00 07 91 00           
[+] >>>> 90 6D 00 00 00           
[+] <<<< 91 1C           
[#] Command code not supported          
[!] ⚠️  Can't get DF Names 

Both chips are DESFire MF3ICD40

iceman1001, 2020-04-28:
try latest,  I added "a fix" based on @bkerler  's PR,   now it doesn't fail, but as @doegox  mentioned we would need a error message  PM3_EUNSUPPORTED_CMD to keep track of things proper.

bkerler, 2020-04-28:
Yes true, it's a quick and dirty fix. Maybe we should verify the version instead and don't run dfname on specific revisions instead.

michal25, 2020-04-29:
Works for Prague Opencard - chip DESFire MF3ICD40

[usb] pm3 --> hf mfdes info
          
[=] --- Tag Information ---------------------------          
[=] -------------------------------------------------------------          
[+]               UID: 04 56 35 E9 76 1B 80           
[+]      Batch number: 8F D4 15 59 60           
[+]   Production date: week 26 / 2008          
          
[=] --- Hardware Information          
[=]      Vendor Id: NXP Semiconductors Germany          
[=]           Type: 0x01          
[=]        Subtype: 0x01          
[=]        Version: 0.2 (DESFire MF3ICD40)          
[=]   Storage size: 0x18 (4096 bytes)          
[=]       Protocol: 0x05 (ISO 14443-2, 14443-3)          
          
[=] --- Software Information          
[=]      Vendor Id: NXP Semiconductors Germany          
[=]           Type: 0x01          
[=]        Subtype: 0x01          
[=]        Version: 0.6          
[=]   Storage size: 0x18 (4096 bytes)          
[=]       Protocol: 0x05 (ISO 14443-3, 14443-4)          
          
[=] --- Card capabilities          
[=]     0.6 - DESFire MF3ICD40, Add ISO/IEC 7816 command set compatibility          
[+]    Number of Masterkeys                  : 1          
[+]    [0x08] Configuration changeable       : YES          
[+]    [0x04] CMK required for create/delete : NO          
[+]    [0x02] Directory list access with CMK : YES          
[+]    [0x01] CMK is changeable              : YES          
[+]    Operation of PICC master key          : (3)DES          
[+]    PICC Master key Version               : 0 (0x00)          
[=]    ----------------------------------------------------------          
[+]    [0x0A] Authenticate      : YES          
[+]    [0x1A] Authenticate ISO  : NO          
[+]    [0xAA] Authenticate AES  : NO          
[=] -------------------------------------------------------------          
          
[=] --- Free memory          
[+]    Card doesn't support 'free mem' cmd          
[=] -------------------------------------------------------------          
[usb] pm3 --> hf mfdes enum
[!] ⚠️  Can't get DF Names          
          
[=] -- Mifare DESFire Enumerate applications --------------------          
[=] -------------------------------------------------------------          
[+]  Tag report 2 applications          
          
[+] --- AMK - Application Master Key settings          
[+]   AID : FFFFFF          
[+]   AID Function Cluster 0xFF: miscellaneous applications          
[+]   AID Key settings           : 0x0b          
[+]   Max key number and type    : 5, (3)DES          
[=] -------------------------------------------------------------          
[+]   Changekey Access rights          
[+]   -- AMK authentication is necessary to change any key (default)          
[+]    [0x08] Configuration changeable       : YES          
[+]    [0x04] AMK required for create/delete : YES          
[+]    [0x02] Directory list access with AMK : NO          
[+]    [0x01] AMK is changeable              : YES          
[=] -------------------------------------------------------------          
[=]   Application keys          
[+]    Key [0]  Version : 0 (0x00)          
[+]    Key [1]  Version : 0 (0x00)          
[+]    Key [2]  Version : 0 (0x00)          
[+]    Key [3]  Version : 0 (0x00)          
[+]    Key [4]  Version : 0 (0x00)          
[+]  Tag report 3 files          
[+]    Fileid 2 (0x02)          
[=]      File Type: 0x00 -> Standard Data File          
[=]      Com.Setting: 0x03 -> Enciphered          
[=]      Access Rights: 0x1200 - Change (Access Key: 0) - RW (Access Key: 0) - W (Access Key: 2) - R (Access Key: 1)          
[=]      Filesize: 48 (0x30)          
[+]    Fileid 1 (0x01)          
[=]      File Type: 0x00 -> Standard Data File          
[=]      Com.Setting: 0x03 -> Enciphered          
[=]      Access Rights: 0x3200 - Change (Access Key: 0) - RW (Access Key: 0) - W (Access Key: 2) - R (Access Key: 3)          
[=]      Filesize: 48 (0x30)          
[+]    Fileid 0 (0x00)          
[=]      File Type: 0x02 -> Value Files with Backup          
[=]      Com.Setting: 0x03 -> Enciphered          
[=]      Access Rights: 0x4200 - Change (Access Key: 0) - RW (Access Key: 0) - W (Access Key: 2) - R (Access Key: 4)          
[=]      Lower limit: 3 (0x3) - Upper limit: 10 (0xA) - limited credit value: 0 (0x0) - limited credit enabled: 0          
          
[+] --- AMK - Application Master Key settings          
[+]   AID : F21150          
[+]   AID Function Cluster 0xF2: reserved          
[+]   AID Key settings           : 0x0b          
[+]   Max key number and type    : 5, (3)DES          
[=] -------------------------------------------------------------          
[+]   Changekey Access rights          
[+]   -- AMK authentication is necessary to change any key (default)          
[+]    [0x08] Configuration changeable       : YES          
[+]    [0x04] AMK required for create/delete : YES          
[+]    [0x02] Directory list access with AMK : NO          
[+]    [0x01] AMK is changeable              : YES          
[=] -------------------------------------------------------------          
[=]   Application keys          
[+]    Key [0]  Version : 0 (0x00)          
[+]    Key [1]  Version : 0 (0x00)          
[+]    Key [2]  Version : 0 (0x00)          
[+]    Key [3]  Version : 0 (0x00)          
[+]    Key [4]  Version : 0 (0x00)          
[+]  Tag report 1 file           
[+]    Fileid 0 (0x00)          
[=]      File Type: 0x01 -> Backup Data File          
[=]      Com.Setting: 0x03 -> Enciphered          
[=]      Access Rights: 0x2010 - Change (Access Key: 0) - RW (Access Key: 1) - W (Access Key: 0) - R (Access Key: 2)          
[=]      Filesize: 480 (0x1E0)          
[=] -------------------------------------------------------------

michal25, 2020-04-29:
But troubles on DESFire EV2
[usb] pm3 --> hf mfdes info
          
[=] --- Tag Information ---------------------------          
[=] -------------------------------------------------------------          
[+]               UID: 04 11 45 92 9F 65 80           
[+]      Batch number: CE EC D3 5D 60           
[+]   Production date: week 21 / 2019          
          
[=] --- Hardware Information          
[=]      Vendor Id: NXP Semiconductors Germany          
[=]           Type: 0x01          
[=]        Subtype: 0x01          
[=]        Version: 12.0 (DESFire EV2)          
[=]   Storage size: 0x18 (4096 bytes)          
[=]       Protocol: 0x05 (ISO 14443-2, 14443-3)          
          
[=] --- Software Information          
[=]      Vendor Id: NXP Semiconductors Germany          
[=]           Type: 0x01          
[=]        Subtype: 0x01          
[=]        Version: 2.1          
[=]   Storage size: 0x18 (4096 bytes)          
[=]       Protocol: 0x05 (ISO 14443-3, 14443-4)          
          
[=] --- Card capabilities          
          
[=] --- Tag Signature          
[=]  IC signature public key name: DESFire EV2          
[=] IC signature public key value: 04B304DC4C615F5326FE9383DDEC9AA8          
[=]                              : 26FE9383DDEC9AA892DF3A57FA7FFB32          
[=]                              : 92DF3A57FA7FFB3276192BC0EAA252ED          
[=]                              : 76192BC0EAA252ED45A865E3B093A3D0          
[=]     Elliptic curve parameters: NID_secp224r1          
[=]              TAG IC Signature: ECFF55858BC813BBB84641F9331952F6          
[=]                              : 9AF0F55F03AB69B4D4EB8A752E981AD7          
[=]                              : 56B5DEEE057960EC92F3A7C8E4A9A3D2          
[=]                              : 84D33CBB129C0A04          
[+]            Signature verified: successful          
[+]    Number of Masterkeys                  : 1          
[+]    [0x08] Configuration changeable       : YES          
[+]    [0x04] CMK required for create/delete : YES          
[+]    [0x02] Directory list access with CMK : YES          
[+]    [0x01] CMK is changeable              : YES          
[+]    Operation of PICC master key          : (3)DES          
[+]    PICC Master key Version               : 0 (0x00)          
[=]    ----------------------------------------------------------          
[+]    [0x0A] Authenticate      : YES          
[+]    [0x1A] Authenticate ISO  : YES          
[+]    [0xAA] Authenticate AES  : NO          
[=] -------------------------------------------------------------          
          
[=] --- Free memory          
[+]    Available free memory on card         : 4896 bytes          
[=] -------------------------------------------------------------          
[usb] pm3 --> hf mfdes enum
          
[=] -- Mifare DESFire Enumerate applications --------------------          
[=] -------------------------------------------------------------          
[+]  Tag report 1 application           
          
[+] --- AMK - Application Master Key settings          
[+]   AID : 123456          
[+]   AID Function Cluster 0x12: reserved          
[!] ⚠️     Can't read Application Master key settings          
[!!] 🚨 APDU: No APDU response.          
[!] ⚠️     Can't read AID master key version. Trying all keys          
[!] ⚠️     Can't get file ids -> Current authentication status does not allow the requested command          
[=] -------------------------------------------------------------

Once more with data setdebugmode 2
[usb] pm3 --> data setdebugmode 2
[usb] pm3 --> hf mfdes enum
[+] >>>> 90 6A 00 00 00           
[+] <<<< 56 34 12 91 00           
[+] >>>> 90 6D 00 00 00           
[+] <<<< 91 00           
          
[=] -- Mifare DESFire Enumerate applications --------------------          
[=] -------------------------------------------------------------          
[+]  Tag report 1 application           
          
[+] --- AMK - Application Master Key settings          
[+]   AID : 123456          
[+]   AID Function Cluster 0x12: reserved          
[+] >>>> 90 5A 00 00 03 56 34 12 00           
[+] <<<< 91 00           
[+] >>>> 90 45 00 00 00           
[+] <<<< 91 AE           
[#] Current authentication status does not allow the requested command          
[!] ⚠️     Can't read Application Master key settings          
[+] >>>> 90 64 00 00 01 00 00           
[!!] 🚨 APDU: No APDU response.          
[#]           
[!] ⚠️     Can't read AID master key version. Trying all keys          
[+] >>>> 90 5A 00 00 03 56 34 12 00           
[+] <<<< 91 00           
[+] >>>> 90 6F 00 00 00           
[+] <<<< 91 AE           
[#] Current authentication status does not allow the requested command          
[!] ⚠️     Can't get file ids -> Current authentication status does not allow the requested command          
[=] -------------------------------------------------------------

bkerler, 2020-04-29:
That seems pretty normal as it indicates that you need to authenticate with the right key first. Authentification isn't yet fully implemented, but will be soon.

iceman1001, 2020-05-20:
@bkerler   You think auth is done that much that we can close this one?

drandreas, 2020-05-26:
I'm a little late to the party. I hope that I'm missing some obvious. How do I execute a combination of auth and read? My assumption (since LED-D stays one) is that auth is persevered and I can simply execute the two commands one after the other, is this correct?

```
[usb] pm3 --> hf mfdes auth -m 1 -t 1 -a 123456 -n 0 -k 123456....
DES selected
[+]   Key        : 12 23 45 78 9A BC DE FF 
[+]   SESSION    : 6A 3E CE 07 CC 84 2F D6 
[=] -------------------------------------------------------------
[=] -------------------------------------------------------------
[usb] pm3 --> hf mfdes readdata -a 123456 -n 00 -l 000000 -t 0
[!] ⚠️     Can't read data -> Current authentication status does not allow the requested command
[!!] 🚨 Couldn't read data. Error -20
```

iceman1001, 2020-06-09:
@drandreas   That is a good question.  I haven't played enough with it.   Maybe @bkerler  has an answer?

closed 2020-06-09 by iceman1001

bkerler, 2020-06-12:
Right now only the authentification itself is implemented. The encryption/decryption after authentification is currently being worked on :)

iceman1001, 2020-06-12:
:)  there has been some changes in the repo since you were here @bkerler

bkerler, 2020-06-12:
Means it's already implemented ? Didn't see the implementation in master ...

iceman1001, 2020-06-12:
No not at all.  Its waiting for you to implement it ;)

drandreas, 2020-06-12:
Thank you for clearing it up. If I find time I might look into it myself.

bkerler, 2020-06-13:
Authentification is now implemented (legacy DES auth was tested successfully) with PR #787