RfidResearchGroup/proxmark3 #593: LTO-7 CM dump size only 8.160 bytes

szolnokit, 2020-03-07:
With LTO-5 cartdridge, I got a 8.160 bytes long binary dump. That's OK.

But I have a problem with LTO-7 CM memory dump.
Following the LTO-7 specification the CM chips size is 16.352 bytes.
But I got only 8.160 bytes long dump.

I see, the block counting loop is fixed size for 255 blocks in "cmdhflto.c". (Every block is 32 bytes long)
But from LTO-6 spec, the total block number is 511 isntead of 255.

Tested with latest spanshot of PM3 (pr v4.0)

Kevin-Nakamoto, 2020-03-08:
This issue is because we do not know 'read block' command that supports > 255 blocks. First of all, "cmdhflot.c" is  based on LTO-1 specification which is public as a part of ECMA standard. According to this specification, read block command shall specify 1 byte of block address to read. This limitation means that we can read data block from 0x00 to 0xFF.  If there is a read block command that can specify "2 byte" of block address, we may be able to read LTO-7 LTO-CM. However, it seems that LTO-7 specification is not public, as far as I know.

See here.
https://www.ecma-international.org/publications/files/ECMA-ST/ECMA-319.pdf

iceman1001, 2020-03-09:
Indeed a problem with not knowing the correct commands for your LTO-7.
If you have a datasheet explaining the commands,  feel free to share it.

iceman1001, 2020-03-09:
Until then I will close this issue.

closed 2020-03-09 by iceman1001