RfidResearchGroup/proxmark3 #601: arm-none-eabi-gcc version 8.3.1 compilation issues

Fl0-0, 2020-03-11:
**Describe the bug**
Since the update of arm-none-eabi-gcc package in Kali linux (gcc 7 to 8) several compilation errors appears.

**Screenshots**

```
mifaresim.c:763:88: error: expected ')' before 'PRIx64'
                     if (DBGLEVEL >= DBG_EXTENDED) Dbprintf("[MFEMUL_WORK] KEY %c: %012" PRIx64, (cardAUTHKEY == 0) ? 'A' : 'B', emlGetKey(cardAUTHSC, cardAUTHKEY));
                                                                                        ^~~~~~~
                                                                                        )
make[1]: *** [../common_arm/Makefile.common:65: obj/mifaresim.o] Error 1

../armsrc/Standalone/hf_young.c: In function 'RunMod':
../armsrc/Standalone/hf_young.c:237:75: error: expected ')' before 'PRIx64'
                         Dbprintf("Simulating ISO14443a tag with uid: %014" PRIx64 " [Bank: %d]", tmpuid, selected);
                                                                           ^~~~~~~
                                                                           )
../armsrc/Standalone/hf_young.c:239:74: error: expected ')' before 'PRIx64'
                         Dbprintf("Simulating ISO14443a tag with uid: %08" PRIx64 " [Bank: %d]", tmpuid, selected);
                                                                          ^~~~~~~
                                                                          )
../armsrc/Standalone/hf_young.c:233:30: error: unused variable 'tmpuid' [-Werror=unused-variable]
                     uint64_t tmpuid = bytes_to_num(uids[selected].uid, uids[selected].uidlen);
                              ^~~~~~
cc1: all warnings being treated as errors
```

**Desktop (please complete the following information):**
 - OS: Kali LInux

iceman1001, 2020-03-11:
Interesting,
PRIx32 compiles but PRIx64 fails.
Its like the 64 bit macros was removed from inttypes...

doegox, 2020-03-13:
I don't understand why this happens but I pushed a fix with a default def for PRIx64 if it happens to not exist

doegox, 2020-03-13:
The issue is also coming with Debian Bullseye

iceman1001, 2020-03-13:
not to mention LF_ICEHID etc...

iceman1001, 2020-03-13:
added two more defines missing,  lf_icehid,  hf_young compiles now aswell.

iceman1001, 2020-03-13:
ok.  closing this.

closed 2020-03-13 by iceman1001

doegox, 2020-03-13:
I tried to follow the rabbit hole, opened https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=953844

iceman1001, 2020-03-14:
Keep falling, brother!