RfidResearchGroup/proxmark3 #602: Error in make clean && make install

mattyams, 2020-03-12:
**Describe the bug**
Installing to a Debian 8 box and getting an error when doing make clean && make install

**To Reproduce**
Steps to reproduce the behavior:
1. cd /proxmark3
2. make clean && make install

**Screenshots**
`[-] CC version.c
In file included from ../include/proxmark3_arm.h:19:0,
                 from version.c:1:
../include/pm3_cmd.h:35:5: error: type of bit-field 'length' is a GCC extension [-Werror=pedantic]
     uint16_t length : 15;  // length of the variable part, 0 if none.
     ^
../include/pm3_cmd.h:79:5: error: type of bit-field 'length' is a GCC extension [-Werror=pedantic]
     uint16_t length : 15;  // length of the variable part, 0 if none.
     ^
cc1: all warnings being treated as errors
../common_arm/Makefile.common:73: recipe for target 'obj/version.o' failed
make[1]: *** [obj/version.o] Error 1
Makefile:91: recipe for target 'bootrom/install' failed
make: *** [bootrom/install] Error 2`

**Desktop (please complete the following information):**
 - OS: Debian 8
 - Can't get other info yet. But it is a Proxmark3 RDV2

**Additional context**
Makefile was changed so that PM3OTHER was selected.
Tried as root and non-root user.
All dependencies are installed.
Restarted multiple times.

iceman1001, 2020-03-12:
You will need to update your GCC toolchain to a later version.

doegox, 2020-03-12:
Debian 8 is the old-old-stable release.
Debian 10 is the current stable release.
Please try a more recent version.

mattyams, 2020-03-12:
That did it, thanks gents.

closed 2020-03-12 by mattyams