RfidResearchGroup/proxmark3 #604: mfdes enum fails getting file ids of applications

eybisi, 2020-03-12:
**mfdes enum fails to get file ids of applications**
I can list file ids manually but mfdes enum doesnt work.

**To Reproduce**

Desfire EV1 card:
```
usb] pm3 --> hf mfdes info
          
-- Desfire Information --------------------------------------          
-------------------------------------------------------------          
  UID                : 04 5D 90 2A E3 40 80           
  Batch number       : BA 64 95 C6 40           
  Production date    : week 25, 2015          
  -----------------------------------------------------------          
  Hardware Information          
      Vendor Id      : NXP Semiconductors Germany          
      Type           : 0x01          
      Subtype        : 0x01          
      Version        : 1.0 (Desfire EV1)          
      Storage size   : 0x18 (4096 bytes)          
      Protocol       : 0x05 (ISO 14443-3, 14443-4)          
  -----------------------------------------------------------          
  Software Information          
      Vendor Id      : NXP Semiconductors Germany          
      Type           : 0x01          
      Subtype        : 0x01          
      Version        : 1.4          
      storage size   : 0x18 (4096 bytes)          
      Protocol       : 0x05 (ISO 14443-3, 14443-4)          
-------------------------------------------------------------          
 CMK - PICC, Card Master Key settings           
          
   [0x08] Configuration changeable       : YES          
   [0x04] CMK required for create/delete : YES          
   [0x02] Directory list access with CMK : NO          
   [0x01] CMK is changeable              : YES          
          
   Max number of keys       : 39          
   Master key Version       : 1 (0x01)          
   ----------------------------------------------------------          
   [0x0A] Authenticate      : YES          
   [0x1A] Authenticate ISO  : YES          
   [0xAA] Authenticate AES  : NO          
          
   ----------------------------------------------------------          
   Available free memory on card       : 1856 bytes          
-------------------------------------------------------------  
```
Manually I can get file ids of application after selecting the app.
```
usb] pm3 --> hf 14a apdu -sk 905a00000342220100
>>>>[sel keep ] 90 5A 00 00 03 42 22 01 00           
<<<< 91 00           
[+] APDU response: 91 00 - OK          
[usb] pm3 --> hf 14a apdu -k 906f000000
>>>>[keep ] 90 6F 00 00 00           
<<<< 01 02 03 04 05 06 07 08 91 00           
[+] APDU response: 91 00 - OK    
```
But when I run `hf mfdes enum` it fails to get file ids of applications
```      
[usb] pm3 --> hf mfdes enum
#db# Can't select card          
Command unsuccessful          
[usb] pm3 --> hf mfdes enum
          
-- Desfire Enumerate Applications ---------------------------          
-------------------------------------------------------------          
 Aid 0 : 42 22 01           
 AMK - Application Master Key settings          
[!]    Can't read Application Master key settings          
[!]    Can't read Application Master key version. Trying all keys          
[!]    Can't get file ids           
[!]    Can't get ISO file ids           
 Aid 1 : 42 22 04           
 AMK - Application Master Key settings          
[!]    Can't read Application Master key settings          
[!]    Can't read Application Master key version. Trying all keys          
[!]    Can't get file ids           
[!]    Can't get ISO file ids           
 Aid 2 : 42 22 05           
 AMK - Application Master Key settings          
[!]    Can't read Application Master key settings          
[!]    Can't read Application Master key version. Trying all keys          
[!]    Can't get file ids           
[!]    Can't get ISO file ids           
 Aid 3 : 42 22 06           
 AMK - Application Master Key settings          
[!]    Can't read Application Master key settings          
[!]    Can't read Application Master key version. Trying all keys          
[!]    Can't get file ids           
[!]    Can't get ISO file ids           
-------------------------------------------------------------  
```

**Desktop:**
 - OS: [Linux]
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]           

 [ CLIENT ]           
  client: RRG/Iceman          
  compiled with GCC 9.2.1 20200130 OS:Linux ARCH:x86_64          

 [ PROXMARK3 RDV4 ]           
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK3 RDV4 Extras ]           
  FPC USART for BT add-on support: absent           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/257a722c 2020-02-11 09:37:24
       os: RRG/Iceman/master/257a722c 2020-02-11 09:44:21
  compiled with GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-01-12 at 15:31: 2
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 279829 bytes (53%) Free: 244459 bytes (47%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```

iceman1001, 2020-03-12:
Yup,   the **hf mfdes** commands could need some love. 
Other issues https://github.com/RfidResearchGroup/proxmark3/issues/589

iceman1001, 2020-03-19:
Try pulling latest and test again.

closed 2020-03-21 by iceman1001