RfidResearchGroup/proxmark3 #613: Different size between compiled files

Windslab, 2020-03-19:
Dear,
This one issue can't be mystake: https://github.com/Gator96100/ProxSpace/issues/32

I would like to ask each member for his shell space to compile this repo code.

1. What is the version of compiller you are use?
2. How are you run a process compilation?
3. What is the settings are you change at the manifests?

iceman1001, 2020-03-19:
As I mentioned over at the issue you linked,  its normal for the different versions of the compilers on the different OS's to generate different sizes of the fullimage.elf.   

What answer is it that you are actually looking for?   What is it that you think is wrong?

Windslab, 2020-03-19:
Proxspace is the same on all windows OS, but size is different. My layout always bigger for 10 kb in avr then from Gator.

It's not native code it can't be different for PLC.  Before I have had worked with SHARK ADSP and never seen like this.

By the way, I need version like Gator use for his builds.

Windslab, 2020-03-19:
All simple - all my builds cannot be used by me. Err of the USB as I wrote before...

iceman1001, 2020-03-19:
no,  there is many versions of Proxspace out.  There is also 32 vs 64 bit distro,  there is also the possibility that a PS has updated its gcc compiler.    It sounds like you have messed up your proxspace distro.

Closing this one,  since its not an issue.

closed 2020-03-19 by iceman1001