RfidResearchGroup/proxmark3 #616: Keri MS Clone with FC and ID

mwalker33, 2020-03-21:
Drafting the code to allow a keri MS clone and supporting the FC and ID

As is :  lf keri clone <internal id> <q5>

Looking for feed back on how you would like the command to be formatted.
in my test code i have
lf keri clone m <fc> <id>
where m = ms. 
if the 'm' option is not supplied it will default back to the current internal id.

Given that we my have other formats in the future would it be better with something like...
lf keri clone t <ms|other types|...> f <fc> i <id> [q5]
and a fallback if no "t" option to the current clone from internal id.
i.e. lf keri clone <internal id> [q5] 

How would you like the command to work ?

Happy to rework as needed.

iceman1001, 2020-03-21:
lets see.    To make sure we gonna get a unify among all lf commands,
when it comes to different types,  I starting to like   MS/INTERNAL  ie long params instead of  1 = ms, 2 = internal.    Where the latter has been used everywhere before but its utterly confusing to keep track of what a 1 means across the commands because of it.

And when using long params,  I tend to convert to str_lower first and then str_startswith,  instead of having at full correct name.   It enables   MS/IN  or ms/in as valid inputs and accepts ms/internal aswell.     Its lax enough to all for wrong spellings...  ie   internalised will also work.  

f  for fc is ok
i for card id,  well,   i suggest    c /cn  for that.  Trying to have that on other lf commands.

mwalker33, 2020-03-21:
Yep, sounds good.
In then sample I added to post 1, as is I was matching the first character (as that's all that's needed (atm)
So lf keri clone t     or lf keri clone type  will be the same
That also extends for the rest
e.g. lf keri clone type ms fc 6 etc...…

I have also left the original format, as I did not want to break something without warning.
I will update the ID to CardID

iceman1001, 2020-03-22:
Done

closed 2020-03-22 by iceman1001