RfidResearchGroup/proxmark3 #618: Bug

Windslab, 2020-03-21:
Err:
is -data.eml

https://github.com/RfidResearchGroup/proxmark3/blob/9a2fc52a9284980b852187221a45d2bc0cc6ee00/client/luascripts/hf-writer.lua#L131

After choice a dump from list it's doesn't write it (can't find patch)

iceman1001, 2020-03-21:
The dump file template for this repo is
**hf-mf-(uid)-dump.(bin/eml/json)**

So I guess you are using dumps from offical repo or old dumps.  I suggest you rename your existing dump files to match current template.

Windslab, 2020-03-21:
No that's from you repo. I play only with it becouse it for human.

Your dumps has a view:
**hf-mf-11223344556677-data.eml**

Yeah probably `dump` will be beter of `data`

iceman1001, 2020-03-21:
bugger, I see...,  its inconsistent implemented in the client.

Alrighty then,  pull latest.  I think I got all places changed now.

Windslab, 2020-03-21:
Okay you changed layout file names.

Script not working to Windows. Seems it something around `find` it is only for bash? Or probably it doesen't see current dir at all.

`Not found: '*dump.eml'`

iceman1001, 2020-03-21:
Just tested it on PS3.2  and it works just fine. 
I don't know what you are doing wrong.

Windslab, 2020-03-21:
[image]

Ofcourse the Eml files inside the dir and names are true...

iceman1001, 2020-03-21:
its waiting for a card,  did you put a card on the pm3?

Windslab, 2020-03-21:
Yes sure. Today afternoon at WSL has been worked. But I have back to Windows.

Windslab, 2020-03-22:
Okay me found a bit problem to Windows platform

Find differents below:
[image]

After modify symbols `""` to `''` it's works but why your code add `----------` prefix to all found files?

[image]

They dumps are found and secusefull printed by `print(_)` but it's lenght is different after prfix that why script printing `Didn't find any dump files`. Could you tell why you have `length = 31` if they should be 29?

Windslab, 2020-03-24:
Iceman are you alive? :)

Ghost-Assassin, 2020-03-29:
> Iceman are you alive? :)

bro, this is rude. Iceman did not owe you anything.

iceman1001, 2020-03-29:
Its ok,  @Windslab  is just very eager.   We have all been that,   its just me who has less time and dont see any need to discuss this particular issue.

Windslab, 2020-03-29:
We are from different countries. I understand your reaction. That's normal.

I not talking like a programist because an engineer. I have fix that script for Windows and now it's doesn't work for Linux. Doing two scripts not a true way. Doing check of the operation system by `if` the same. As my point you can't leave it as now. That's why my asking a code optimization.

Many Thanks

iceman1001, 2020-03-29:
Problem is that my solution works on linux,  osx, wsl2 and proxspace 3.2,  and you seem to belive your solution is better.  Hence I don't feel the need to discuss this issue any more.  For me this is a non-issue until you prove your solution is the better one.

Windslab, 2020-03-29:
Seems we talking about different things. As I have written that's normal.

The script from your repository are not working on my machine, at all. My machine has Cyrillic language maybe that's why different results we are have. As you see I open for conversation to understend what's going on. 

Could you check it please? How I can help?
I assume possibly the answer is find files by prefix `hf-mf-`UID`-dump.eml` in list that is returned by a **find** command. Possible it is can be do more easey, like this `-`UID`-`.

Purpose of my `Pull` are to show for you differents in script, I even not hope to see it in the repo.

closed 2020-04-15 by Windslab