RfidResearchGroup/proxmark3 #619: Incorrect key emulation

Akisame-AI, 2020-03-21:
I have moved to a new apartment and this one uses simple MFC 1k keyfobs for entering. All it checks is the UID, sector 0 key and sector 1 block 0 and key. I have extracted all the keys via a hardnested attack and dumped the card completely. When writing this dump to a direct write or OTW card I can open the doors (when I write it to a magic card the readers do not respond at all). 
However when I try to simulate this dump the doors do not react at all.
A reader attack showed no response from the readers. 
Sniffing with the PM3 in between the card and the reader showed no reaction from the reader.
When I used mifare classic tools to read it using the actual keys it showed: "None of the keys were valid for reading"

I am running the following commands

hf mf eclr
hf mf eload hf-mf-<uid>-data
hf mf sim

when trying to do a reader attack I get nothing either

`[usb] pm3 --> hf mf sim u xxxxxxxx i x e v                                                                              [=] Mifare   | 4 byte UID  AD 78 1C F8                                                                                  [=] Options [ numreads: 0, flags: 35 (0x23) ]                                                                           [=] Press pm3-button or send another cmd to abort simulation                                                            #db# 4B UID: ad781cf8                                                                                                   #db# ATQA  : 00 04                                                                                                      #db# SAK   : 08                                                                                                         #db# Emulator stopped. Tracing: 1  trace length: 0                                                                      |---|----------------|---|----------------|---|                                                                         |sec| key A          |res| key B          |res|                                                                         |---|----------------|---|----------------|---|                                                                         |000|  ------------  | 0 |  ------------  | 0 |                                                                         |001|  ------------  | 0 |  ------------  | 0 |                                                                         |002|  ------------  | 0 |  ------------  | 0 |                                                                         |003|  ------------  | 0 |  ------------  | 0 |                                                                         |004|  ------------  | 0 |  ------------  | 0 |                                                                         |005|  ------------  | 0 |  ------------  | 0 |                                                                         |006|  ------------  | 0 |  ------------  | 0 |                                                                         |007|  ------------  | 0 |  ------------  | 0 |                                                                         |008|  ------------  | 0 |  ------------  | 0 |                                                                         |009|  ------------  | 0 |  ------------  | 0 |                                                                         |010|  ------------  | 0 |  ------------  | 0 |                                                                         |011|  ------------  | 0 |  ------------  | 0 |                                                                         |012|  ------------  | 0 |  ------------  | 0 |                                                                         |013|  ------------  | 0 |  ------------  | 0 |                                                                         |014|  ------------  | 0 |  ------------  | 0 |                                                                         |015|  ------------  | 0 |  ------------  | 0 |                                                                         |016|  ------------  | 0 |  ------------  | 0 |                                                                         |017|  ------------  | 0 |  ------------  | 0 |                                                                         |018|  ------------  | 0 |  ------------  | 0 |                                                                         |019|  ------------  | 0 |  ------------  | 0 |                                                                         |020|  ------------  | 0 |  ------------  | 0 |                                                                         |021|  ------------  | 0 |  ------------  | 0 |                                                                         |022|  ------------  | 0 |  ------------  | 0 |                                                                         |023|  ------------  | 0 |  ------------  | 0 |                                                                         |024|  ------------  | 0 |  ------------  | 0 |                                                                         |025|  ------------  | 0 |  ------------  | 0 |                                                                         |026|  ------------  | 0 |  ------------  | 0 |                                                                         |027|  ------------  | 0 |  ------------  | 0 |                                                                         |028|  ------------  | 0 |  ------------  | 0 |                                                                         |029|  ------------  | 0 |  ------------  | 0 |                                                                         |030|  ------------  | 0 |  ------------  | 0 |                                                                         |031|  ------------  | 0 |  ------------  | 0 |                                                                         |032|  ------------  | 0 |  ------------  | 0 |                                                                         |033|  ------------  | 0 |  ------------  | 0 |                                                                         |034|  ------------  | 0 |  ------------  | 0 |                                                                         |035|  ------------  | 0 |  ------------  | 0 |                                                                         |036|  ------------  | 0 |  ------------  | 0 |                                                                         |037|  ------------  | 0 |  ------------  | 0 |                                                                         |038|  ------------  | 0 |  ------------  | 0 |                                                                         |039|  ------------  | 0 |  ------------  | 0 |                                                                         |---|----------------|---|----------------|---|                                                                         [=] ( 0 :Failed / 1 :Success)           `

Sniffing with a valid tag in between only shows me this

  `97093740 |   97094796 | Rdr |26                                                                       |     | REQA`
as well as the tag's response (its UID with an extra 31 at the end)
which looks like it is timeing out. (it explains why it took so long to authenticate)

 hf 14a sim t 1 u AD781CF8 x e v
did nothing with the reader. MCT showed it used a default key to try to authenticate (a0a1a2a3a4a5).

I also tried "hf 14A sim" but to no avail.

Can someone tell me what I am doing wrong?

`[usb] pm3 --> hw version                                                                                                #db# Emulator stopped. Tracing: 1  trace length: 19880                                                                                                                                                                                           [ Proxmark3 RFID instrument ]                                                                                                                                                                                                                   [ CLIENT ]                                                                                                               client: RRG/Iceman                                                      [... cut ...]  [+] HF antenna is OK                                                                                                                                                                                                                            [+] Displaying LF tuning graph. Divisor 88 is 134.83 kHz, 95 is 125.00 kHz.                                                                                                                                                                     [usb] pm3 -->     `

iceman1001, 2020-03-22:
It seem like you are having problems not knowing how to run the commands,  rather than reporting a issue / bug with the source code. 

We use the Proxmark3 forum ( http://www.proxmark.org/forum/index.php )   for those purposes.

Akisame-AI, 2020-03-22:
Thanks. Sorry for the trouble

closed 2020-03-22 by Akisame-AI