RfidResearchGroup/proxmark3 #627: hf 14a apdu: invalid option "- "

Ghost-Assassin, 2020-03-28:
For Git version: https://github.com/RfidResearchGroup/proxmark3/commit/efeff2d0f68e209597e8dae07c70b029cbdf156f

**To Reproduce**
1. git clone https://github.com/RfidResearchGroup/proxmark3.git
2. cd proxmark3
3. make clean
4. make PLATFORM=PM3OTHER
5. cd client
6. sudo ./proxmark3 /dev/ttyACM0

When running commands:

`hf 14a apdu -st 00A404000E325041592E5359532E444446303100` 
or any other commands shown in the "Sample".

The hf 14a apdu feature is not working correctly, shown in the screenshot. The apdu message was not sent, and complained on some not existing "- ". 
[image]

**Desktop (please complete the following information):**
 - OS: Tried on Ubuntu 16.04, Ubuntu 18.04, Kali 2019.4
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version (shown above)
 - hw status
[image]

 - data tune (LF antenna was removed purposely)
[image]

Thanks in advance

Ghost-Assassin, 2020-03-29:
Seems other commands may also have this problem. 

[image]

iceman1001, 2020-03-29:
Try pulling latest and try again

Ghost-Assassin, 2020-03-29:
WORK LIKE A CHARM!!! Thank you!

closed 2020-03-29 by Ghost-Assassin