RfidResearchGroup/proxmark3 #628: [idea] Chain hf 14a sim with send raw or send apdu

Ghost-Assassin, 2020-03-29:
**Is your feature request related to a problem? Please describe.**

This is a feature request. 
We tried to attack a reader by simulating the pm3 as a JCOP card. 
The current send raw, send apdu are designed to spoof the card, not card reader. 

**Describe the solution you'd like**
If it is possible, I wish this could be added to hf 14a sim, or a new feature combined "sim" and "apdu" which can make the PM3 be able to automatically answer "Select Card" "Select AID" or other APDU messages.

For example, when the reader asked for "Select AID", PM3 should be able to respond to the card reader with our custom message. Consider like searching a dictionary or a JSON file or JSON string and send the result back to card reader and wait for the next APDU request.

iceman1001, 2020-03-29:
That was a very interesting idea.

merlokk, 2020-03-29:
it was here)
https://github.com/merlokk/proxmark3/commits/EMVemulator

but it works not stable because of usb communication model in 2018

iceman1001, 2020-03-29:
You should finish it @merlokk   :)

merlokk, 2020-03-29:
now the most requested thing is the time(
i plain to finish this)

iceman1001, 2020-03-29:
no lockup for corona?   Maybe @Ghost-Assassin  will insist

Ghost-Assassin, 2020-03-29:
This feature would be really helpful for a project I am currently working on. 

I just checked the branch "EMVemulator" which is not compiling. Not sure if I could just ignore the errors or further investigation need to be done. 
[image]

I am currently using a workaround to accomplish this idea. I used PM3 to sniff the conversation between JCOP card and the reader, then use an Android phone with the Xposed Framework as an AID router, which will forward the APDU message to another NFC app. This app will process the request and give a response by an exact match or a pattern match. Just in case anyone got interested, I left an Android Studio project and a debug apk here. https://github.com/xuanxuanblingbling/NFChacker/issues/3

iceman1001, 2020-03-29:
that branch looks like it needs to pull latest source from upstream.