RfidResearchGroup/proxmark3 #637: command execution time out

cosmo6110, 2020-04-03:
lf search 
EM410x tag, get "**command execution time out**", 

[usb] pm3 --> lf search
[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...
[=]
[+] EM410x pattern found
//*************
********//
[+] Valid EM410x ID found!

[!] command execution time out

[usb] pm3 --> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........39944
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............0
#db# Currently loaded FPGA image
#db#   mode.................... HF image built for 2s30vq100 on 2020-01-12 at 15:31:16
#db# Flash memory
#db#   Baudrate................24 MHz
#db#   Init....................OK
#db#   Memory size.............2 mbits / 256 kb
#db#   Unique ID...............0xD5690C23DF9F262A
#db# Smart card module (ISO 7816)
#db#   version.................v3.10
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125.00 kHz )
#db#   [b] bits per sample.....8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db#   [s] samples to skip.....0
#db# LF T55XX config
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
#db#            mode            |start|write|write|write| read|write|write
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
#db#     long leading reference |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
#db#               leading zero |  29 |  17 |  15 |  40 |  15 | N/A | N/A |
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
#db#
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............500ms
#db#   Bytes transferred.......166400
#db#   Transfer Speed PM3 -> Client = 332800 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db#   Slow clock..............31449 Hz
#db# Installed StandAlone Mode
#db#   LF EM4100 read/write/clone mode
#db# Flash memory dictionary loaded
#db#   Mifare..................906 keys
#db#   T55x7...................109 keys
#db#   iClass..................7 keys
[usb] pm3 -->

[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman
  compiled with MinGW-w64 9.2.0 OS:Windows (64b) ARCH:x86_64

 [ PROXMARK3 RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK3 RDV4 Extras ]
  FPC USART for BT add-on support: present

 [ ARM ]
  bootrom: RRG/Iceman/master/731e711e-dirty-unclean 2020-04-03 11:45:53
       os: RRG/Iceman/master/731e711e-dirty-unclean 2020-04-03 11:47:07
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-02-22 at 12:51:14
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 281917 bytes (54%) Free: 242371 bytes (46%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

iceman1001, 2020-04-03:
Yeah, I am aware of it.

mwalker33, 2020-04-05:
I am seeing the same issue on the rdv4 (win10) as well.
e.g.
a write command works.
a non search read seams to work (e.g. lf em 410x_read, lf t55 detect both OK)
but lf search will find the card, but then timeout leaving led A on.
Happy to have a look if you have an idea where the issue may be.

mwalker33, 2020-04-05:
Found it.  Fix in pull request.

blissjoe, 2020-04-05:
That fix worked for me. Thank you.

iceman1001, 2020-04-05:
should fix it::  #645

closed 2020-04-05 by iceman1001