RfidResearchGroup/proxmark3 #642: Unable to run client

PhoenixCampbell, 2020-04-04:
While compiling, I am unable to run the client at the step indicated:

pm3 ~/client$ pm3 /dev/ttyACM0
[=] Waiting for Proxmark3 to appear...
[=] Session log C:\Users\PHJCa\OneDrive\Documents\GitHub\ProxSpace-3.3\pm3/.proxmark3/log_20200404.txt
[!!] ERROR: cannot parse command line. We got COM3  as port and now we got also: C:/Users/PHJCa/OneDrive/Documents/GitHub/ProxSpace-3.3/msys2/dev/ttyACM0

syntax: proxmark3.exe [-h|-t|-m]
        proxmark3.exe [[-p] <port>] [-b] [-w] [-f] [-c <command>]|[-l <lua_script_file>]|[-s <cmd_script_file>] [-i] [-d <0|1|2>]
        proxmark3.exe [-p] <port> --flash [--unlock-bootloader] [--image <imagefile>]+ [-w] [-f] [-d <0|1|2>]

pm3 ~/client$ proxmark3 /dev/ttyACM0
[=] Session log C:\Users\PHJCa\OneDrive\Documents\GitHub\ProxSpace-3.3\pm3/.proxmark3/log_20200404.txt
[=] Using UART port C:/Users/PHJCa/OneDrive/Documents/GitHub/ProxSpace-3.3/msys2/dev/ttyACM0

[!] ERROR: invalid serial port C:/Users/PHJCa/OneDrive/Documents/GitHub/ProxSpace-3.3/msys2/dev/ttyACM0

I was wondering what I am doing wrong. I was also following along with the video installation as well, wondering if that led to problems as well as written direction.

iceman1001, 2020-04-04:
either you use the scripts which tries to find which serial port you are on.
```
./pm3
```
or if you run it from root folder, 
```
./client/proxmark3.exe -p /dev/ttyACM0
```

PhoenixCampbell, 2020-04-04:
i got it to work, one last question, how do you switch from offline to online mode. That seems to be my last problem.

iceman1001, 2020-04-04:
Good that you solved your issue.

And sorry, I don't understand that question.  
If you are asking about standalone mode,  there is plenty of documents here in the repo that covers that part.

closed 2020-04-04 by iceman1001