RfidResearchGroup/proxmark3 #643: Cannot communicate with Proxmark

PhoenixCampbell, 2020-04-04:
**Describe the bug**
I am unable to communicate with my proxmark tool. I have it in, all drivers are installed and the port shows a proxmark3 (com3)

Here are my lines that show I can run in "offline" mode but can't talk to the tool itself.

pm3 ~$ client/proxmark3
[=] Session log C:\Users\PHJCa\OneDrive\Documents\GitHub\ProxSpace-3.3\pm3/.proxmark3/log_20200404.txt
[=] Running in OFFLINE mode. Check "proxmark3.exe -h"  if it's not what you want.

  ██████╗ ███╗   ███╗ ████╗
  ██╔══██╗████╗ ████║   ══█║
  ██████╔╝██╔████╔██║ ████╔╝
  ██╔═══╝ ██║╚██╔╝██║   ══█║       iceman@icesql.net
  ██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/
  ╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0

[offline] pm3 --> exit
pm3 ~$ pm3
[=] Waiting for Proxmark3 to appear...
[=] Session log C:\Users\PHJCa\OneDrive\Documents\GitHub\ProxSpace-3.3\pm3/.proxmark3/log_20200404.txt
[=] Using UART port COM3
unknown command:: 0x61334d50[!!] ERROR: cannot communicate with the Proxmark

This is on Windows 10.

iceman1001, 2020-04-04:
you seem to have forgotten to flash your device,  it seems like you should benefit from reading the documents in the repo.

closed 2020-04-04 by iceman1001

iBobik, 2021-03-05:
@iceman1001 I've read [README](https://github.com/RfidResearchGroup/proxmark3/blob/master/README.md), but there is no info about need of flashing nor link to a document describing flash process.

iceman1001, 2021-03-05:
Try harder.

iBobik, 2021-03-05:
I am probably not good enought to use your SW. :-(