RfidResearchGroup/proxmark3 #646: get_desfire_fileids() fails

ikarus23, 2020-04-05:
The `get_desfire_fileids()` function as part of `hf mfdes enum` fails. The output just shows "Command unsuccessful". It should work because NXP's TagInfo is sowing the files. I don't think there is something special about the tag I'm using.

ikarus23, 2020-04-05:
After the latest code changes to DESFire, the error is now `[SendDesfireCmd] Unsuccessful: 6F 00 00 00`

iceman1001, 2020-04-05:
Could be that TagInfo authenticates with a default key before.

ikarus23, 2020-04-05:
I don't think so. The tag has no default key configured in any app.

iceman1001, 2020-04-06:
try pulling latest and see if it becomes better

ikarus23, 2020-04-06:
Well... it changed! It seams like it worked on the first app, but not on the second.
```
...

[=]   Application keys          
[+]    Key [0]  Version : 0 (0x00)          
[+]    Key [1]  Version : 0 (0x00)          
[+]  Tag report 5 files          
[+]    Fileid 0 (0x00)          
[=]   Settings [7] 00 01 00 10 20 00 00          
[+]    Fileid 1 (0x01)          
[=]   Settings [13] 04 00 00 10 08 00 00 3D 00 00 00 00 00          
[+]    Fileid 2 (0x02)          
[=]   Settings [7] 00 01 00 10 20 01 00          
[+]    Fileid 8 (0x08)          
[=]   Settings [7] 01 00 00 E0 20 00 00          
[+]    Fileid 3 (0x03)          
[+]  AID 3 : XX XX XX           
          
[+] --- AMK - Application Master Key settings           
#db# Can't select card          
[!]  ⚠️  [SendDesfireCmd] Unsuccessful: 5A 00 00 03 D0 82 F4 00           
[!]  ⚠️  [get_desfire_select_application] Can't select AID: D0 82 F4           
[!]  ⚠️     Can't select AID           
[+]  Tag report 3 files          
[+]    Fileid 1 (0x01)          
[+]    Fileid 6 (0x06)          
[=]   Settings [7] 00 03 F1 23 10 00 00          
[+]    Fileid 4 (0x04)          
[=] ------------------------------------------------------------- 
```

iceman1001, 2020-04-06:
One step in the right direction then :)

bkerler, 2020-04-07:
I think I've sorted the problem. The field must be deselected before each select application id call. I've pushed a commit 61f3ef6.

ikarus23, 2020-04-07:
Yes! It works now! Thank you so much!

Maybe another issue or it is me not understanding this output, but why are having two of the three files no settings?
```
[+]  Tag report 3 files          
[+]    Fileid 1 (0x01)          
[+]    Fileid 6 (0x06)          
[=]   Settings [7] 00 03 F1 23 10 00 00          
[+]    Fileid 4 (0x04)  
```

bkerler, 2020-04-07:
I don't think its a bug. But to be sure, please add the "hf mfdes list" command after running "hf mfdes enum" :)

ikarus23, 2020-04-07:
Maybe "bug" is the wrong word here. I just don't understand, why there are not settings for each file. Judging by the output of NXP's TagInfo, each file has its settings. The PM3 output shows 3 files, but only one Settings entry. Am I missing something?

The tag responses to all 3 `GET FILE SETTINGS`, but not all answers are shown. Whats wrong with the other answers?
 
```          0 |        992 | Rdr |52                                                                       |     | WUPA          
       2100 |       4468 | Tag |44  03                                                                   |     |           
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL          
      10548 |      16372 | Tag |88  04  69  64  81                                                       |     |           
      19200 |      29728 | Rdr |93  70  88  04  69  64  81  e2  75                                       |  ok | SELECT_UID          
      30772 |      34292 | Tag |24  d8  36                                                               |     |           
      35712 |      38176 | Rdr |95  20                                                                   |     | ANTICOLL-2          
      39220 |      45044 | Tag |fa  dc  5a  80  fc                                                       |     |           
      47744 |      58272 | Rdr |95  70  fa  dc  5a  80  fc  a5  7c                                       |  ok | SELECT_UID-2          
      59316 |      62900 | Tag |20  fc  70                                                               |     |           
      64640 |      69408 | Rdr |e0  80  31  73                                                           |  ok | RATS          
      70452 |      79732 | Tag |06  75  77  81  02  80  02  f0                                           |  ok |           
      87936 |     101856 | Rdr |02  90  5a  00  00  03  d0  82  f4  00  25  72                           |  ok | SELECT APPLICATION          
     123060 |     128948 | Tag |02  91  00  29  10                                                       |     |           
     411136 |     420448 | Rdr |03  90  6f  00  00  00  c6  2e                                           |  ok | GET FILE IDS          
     426804 |     436084 | Tag |03  01  06  04  91  00  f4  18                                           |  ok |           
     718720 |     730400 | Rdr |02  90  f5  00  00  01  00  00  79  e3                                   |  ok | GET FILE SETTINGS          
     737204 |     743028 | Tag |02  91  f0  a6  e7                                                       |     |           
    1027840 |    1039456 | Rdr |03  90  f5  00  00  01  01  00  1e  7b                                   |  ok | GET FILE SETTINGS          
    1044916 |    1058868 | Tag |03  00  03  f1  23  10  00  00  91  00  1f  85                           |  ok |           
    1341824 |    1353504 | Rdr |02  90  f5  00  00  01  02  00  c9  d0                                   |  ok | GET FILE SETTINGS          
    1360308 |    1366132 | Tag |02  91  f0  a6  e7                                                       |     |  
```

iceman1001, 2020-04-08:
Need to look up this reposons code
0x91  f0

iceman1001, 2020-04-08:
0xF0 - file not found.
Interesting.   So the return file_ids collected before isn't correct.

ikarus23, 2020-04-08:
Well, the IDs 0x01, 0x06 and 0x04 are correct (confirmed by TagInfo).

ikarus23, 2020-04-10:
The commands no cause further issues. Both, `info` and `enum` are no longer able to select the AID.
```
[usb] pm3 --> hf mfdes info
          
[=] --- Tag Information ---------------------------          
[=] -------------------------------------------------------------          
[+]               UID: 04 69 64 FA DC 5A 80            
[+]      Batch number: B9 0C 17 55 20            
[+]   Production date: week 52 / 2017           
          
[=] --- Hardware Information           
[=]      Vendor Id: NXP Semiconductors Germany           
[=]           Type: 0x01           
[=]        Subtype: 0x01           
[=]        Version: 1.0 ( DESFire EV1 )          
[=]   Storage size: 0x1A ( 8192 bytes )          
[=]       Protocol: 0x05 ( ISO 14443-2, 14443-3 )          
          
[=] --- Software Information           
[=]      Vendor Id: NXP Semiconductors Germany           
[=]           Type: 0x01           
[=]        Subtype: 0x01           
[=]        Version: 1.4           
[=]   Storage size: 0x1A ( 8192 bytes )          
[=]       Protocol: 0x05 ( ISO 14443-3, 14443-4 )          
          
[=] --- Card capabilities           
[=] 	1.4 - DESFire Ev1 MF3ICD21/41/81, EAL4+, N/A (report to iceman!)          
[!]  ⚠️     Can't select AID 0x0 -> Invalid argument(s)           
          
[=] --- Free memory           
[+]    Available free memory on card         : 4032 bytes           
[=] -------------------------------------------------------------          
[usb] pm3 --> hf mfdes enum
          
[=] -- Mifare DESFire Enumerate applications --------------------          
[=] -------------------------------------------------------------          
[+]  Tag report 2 applications          
          
[+] --- AMK - Application Master Key settings           
[+]   AID : XX XX XX           
[!]  ⚠️     Can't select AID 0xXXXX -> Invalid argument(s) 
```

ikarus23, 2020-04-10:
Is this because the leading zeros of AIDs get stripped (at least in the output)?

bkerler, 2020-04-11:
Desfire is currently WIP. Please test the latest commit, it should fix several issues. Make sure to also flash your pm3.

ikarus23, 2020-04-11:
@bkerler Wow! This is great! Thanks for supporting @iceman1001 and this project with your knowledge about DESFire!

I've checked the last DESFire commits and a lot of things work great. Some issue however still remain:
```
[+]  Tag report 3 files          
[+]    Fileid 1 (0x01)          
[!]  ⚠️     Can't get file settings -> Specified file number does not exist           
[+]    Fileid 6 (0x06)          
[!!]  🚨 APDU: No APDU response.          
[!]  ⚠️     Can't get file settings ->            
[+]    Fileid 4 (0x04)          
[!!]  🚨 APDU: No APDU response.          
[!]  ⚠️     Can't get file settings -> 
```

bkerler, 2020-04-11:
Can you do a "data setdebugmode 2" and then rerun the command and upload the log please ?

ikarus23, 2020-04-11:
Sure thing!
```
[+] >>>> 90 6F 00 00 00           
[+] <<<< 01 06 04 91 00           
[+]  Tag report 3 files          
[+]    Fileid 1 (0x01)          
[+] >>>> 90 F5 00 00 01 00 00           
[+] <<<< 91 F0           
[#] Specified file number does not exist          
[!]  ⚠️     Can't get file settings -> Specified file number does not exist           
[+]    Fileid 6 (0x06)          
[+] >>>> 90 F5 00 00 01 01 00           
[!!]  🚨 APDU: No APDU response.          
[#]           
[!]  ⚠️     Can't get file settings ->            
[+]    Fileid 4 (0x04)          
[+] >>>> 90 F5 00 00 01 02 00           
[!!]  🚨 APDU: No APDU response.          
[#]           
[!]  ⚠️     Can't get file settings ->            
[=] -------------------------------------------------------------  
```

bkerler, 2020-04-11:
Found and hopefully fixed the bug in pull request #669 :)

iceman1001, 2020-04-11:
Yes,  I'd like to express my gratitude aswell.

ikarus23, 2020-04-11:
Yes! The bug is fixed! Thank you.

Just another thing I've noticed: `hf mfdes info` gives me detailed information about the key configuration of AID 0. The details of `hf mfdes enum` on key configuration of the apps is not that detailed. e.g. It misses the key type ((3)DES, 2/3kDES, AES).

bkerler, 2020-04-12:
The key type is only shown for the PICC. There is no command to get the key type for the card level (master keys) for the aids (that's what the datasheet says).

ikarus23, 2020-04-13:
Hmm I guess there must be such a command, because NXP's TagInfo displays me this information. For each app on the card there is a section with e.g. "2 (3)DES keys" or "14 AES keys".

iceman1001, 2020-04-13:
some screen shots?   When I use NXP TagInfo app for IOS ,  it doesn't show me anything if its not NDEF....

ikarus23, 2020-04-13:
Here are two screenshot of two apps on a card I have laying around.
[image]

[image]

(@iceman1001: Can't imagine being a NFC/RFID nerd without and Android device at hand ;) )

iceman1001, 2020-04-13:
I am not an Android lover.    With that said, I do actually have an Android layin' around which I somehow need to flash all firmware to get all chinese stuff out of it.

ikarus23, 2020-04-13:
Yeah, I know. Me neither. But I don't love iOS instead. Just two operating systems with their pros and cons. But Android of cheap Chinese phones != "real" Android. Well, in any case, Android is good tool for NFC, but you know that already ;)  And this is going a bit off topic :P So enough of that.

bkerler, 2020-04-13:
Ok, thanks for the input. If auth is being used for each aid, it might be possible to detect. At least in the datasheet I couldn't see anything like that. Will have a closer look at that :)

bkerler, 2020-04-13:
They seem to use getkeysettings for that. Although it should only work for the picc master key as stated in the datasheet, it seems to work for application master keys as well 8-]. Feature was added in pull request #680.

iceman1001, 2020-04-14:
seems its hard to know who is right.  TagInfo app or the datasheet ....

ikarus23, 2020-04-14:
@bkerler Thanks again for all the effort! It works now. And other commands seem to work too (read a standard file).

Side notice: the command syntax is unusual for the PM3 project. For most commands the arguments don't use the `-`. So `hf mf sim h` vs. `hf mfdes readdata -h`. `hf mfdes readdata h` does not work. Don't get me wrong, I like the `-option` syntax. It's feels more like Linux. But it is rarely used in PM3. But since the PM3 is inconsistent at a lot of places I wouldn't bother. ;)

Thanks again you two for making the PM3 a DESFire compatible tool!

closed 2020-04-14 by ikarus23

iceman1001, 2020-04-15:
Complete sidenote

... yes and no..   The old style used no dash,  only short params. 
Then came the raw cmd, which used dash for short params,  
then came lua which uses dash for short params
then came cliparser ( issue on offical repo for better param parsing since it wasn't good enough, https://github.com/Proxmark/proxmark3/issues/467)
which uses  dash (versal/capital) short, double dash w long param,  and that is where we are today.

to old mess is still there.  Swapping out to cliparser is a job ppl isn't prepared to do.
but for new stuff,  where we need many different params and long hex strings,  the cliparser works best.  Can't use the old ways.

So this is just the normal growing pains for the proxmark client.  Living in three worlds with regards to cli parameter parsing :)