RfidResearchGroup/proxmark3 #648: Magic chinese gen3 support

0ki, 2020-04-05:
So there are these "gen3" S70 (4k) 7bit UID cards on the market (I think 4bit versions also exist) that use different chinese magic.
Seems that all of them come with UID 041219C3219316 ( "04 12 19 C3 21 93 16" ) from the factory.

As per @iceman1001 UID can be changed as follows:
`hf 14a raw -s -c -t 2000 90f0cccc10PPQQRRSSTTUUVV984200e32000000011`
http://www.proxmark.org/forum/viewtopic.php?id=6530
http://www.proxmark.org/forum/viewtopic.php?id=6545

This works and `hf search` confirms the new UID, however sector 0 remains the same. Is that expected behaviour?

What about having some nice commands for gen3 support?

P.S. There is also lock UID function: 90fd111100

iceman1001, 2020-04-06:
These new Gen3 styled cards,  its a mess.  Many different chinese suppliers make their own styled and what exactly their cards capabilities isn't in any orderly manor defined.  Its just "lets make a card" kind of attitude.  

So if uid isn't reflected onto block0 or not,  I don't know.
We only know that you need to send custom apdu's to the card to change uid,  and  perma lock it.

I do agree, there should be a lua-script for these types.

Windslab, 2020-04-22:
> What about having some nice commands for gen3 support?

You can find this functionality at `mfc_gen3_writer.lua`

Close.

closed 2020-04-23 by iceman1001