RfidResearchGroup/proxmark3 #652: [idea] Mifare DESFire dump

Sympatron, 2020-04-06:
Since recently the `hf mfdes enum` command is working and it is quite helpful for a basic overview of the tags content.

It would be even more useful to have a very basic dump of all accessible files. I.e. listing all files of all applications allowing that without authentication and dumping the content if possible.

iceman1001, 2020-04-06:
Full support for DESFire with extra lika dump would be awesome.  Its an old wish that ppl has been talking about since before 2013.  
Do feel free to contribute!

Sympatron, 2020-04-07:
I just looked into it and it seems `mfdes enum` already does most of it. Just dumping file contents is missing.

I did not realize this, because listing the files still does not work for me. After printing the key versions I get an error:
```
[+] --- AMK - Application Master Key settings
[+]   AID Key settings           : ef
[+]   Max number of keys in AID  : 131
[=] -------------------------------------------------------------
[+]   Changekey Access rights
[+]   -- Authentication with the key to be changed (same KeyNo) is necessary to change a key
[+]    [0x08] Configuration changeable       : YES
[+]    [0x04] AMK required for create/delete : NO
[+]    [0x02] Directory list access with AMK : NO
[+]    [0x01] AMK is changeable              : YES
[=] -------------------------------------------------------------
[=]   Application keys
[+]    Key [0]  Version : 0 (0x00)
[+]    Key [1]  Version : 0 (0x00)
[+]    Key [2]  Version : 0 (0x00)
[!] [SendDesfireCmd] Unsuccessful: 6F 00 00 00
[=] -------------------------------------------------------------
```

I don't see what's wrong, because selecting the application seems to have worked (otherwise I would have gotten a `Can't select AID` error) and 6F seems to be the correct command and authentication is not necessary to list the files in this application.

bkerler, 2020-04-08:
Do you have the latest commit ? This error shouldn't happen anymore.

Sympatron, 2020-04-08:
I pulled the latest code 2 days ago, compiled and flashed the new firmware. After that `mfdes enum` worked better but still produced this error.

iceman1001, 2020-04-08:
Try again.

Sympatron, 2020-04-09:
Now it works :+1:

bkerler, 2020-04-11:
Regarding your initial request for "dumping" Mifare DESFire ... this is currently WIP.

bkerler, 2020-04-13:
Just added read/write functionality and other stuff in pull request #680, which should conclude with your issue request.

iceman1001, 2020-04-14:
i think  a  "hf mfdes dump"  would mean try to enumerate all aids,  and download all files and save on client.   Don't think its gonna work given all kinds of keys that are possible for each AID.

bkerler, 2020-04-14:
We could try to do that for the factory keys at least ;)

iceman1001, 2020-04-14:
yes,  I haven't seen any setting that keeps track of failed authentication attempts.  So that venue is open.
Which also leads to a  **hf mfdes chk**   command ,  where we take a default / known key list ...

iceman1001, 2020-05-20:
Closing this one,  since its too wage described to what kind of solution that would fit Desfire with its directory/file based structure. 

The current `hf mfdes *` implemenation gives the possibilty to identify and read all files if you have the right key or file is unprotected.

Better to start a new issue with specified details on how such a command should be implemented.
No more _wage_ wishlist on the repo :)

closed 2020-05-20 by iceman1001