RfidResearchGroup/proxmark3 #659: Braces Error When Compiling cmdhfmfdes.c on Mac OSX Mojave/Catalina

capnkrunchy, 2020-04-08:
**Describe the bug**
When compiling any commit newer than 7594304 on Mac OSX, `make all` throws the following error(s):
`[-] CC cmdhfmfdes.c`
`cmdhfmfdes.c:713:30: error: suggest braces around initialization of subobject`
`      [-Werror,-Wmissing-braces]`
`    dfname_t dfnames[255] = {0};`
......
`cmdhfmfdes.c:886:30: error: suggest braces around initialization of subobject`
 `     [-Werror,-Wmissing-braces]`
 `   dfname_t dfnames[255] = {0};`

**To Reproduce**
Steps to reproduce the behavior:
1. Pull latest version or any version after 7594304
2. `make clean && make all`

**Expected behavior**
Compile without errors

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Mac OSX 10.14.6 and Mac OSX 10.15.4
 - Output of clang --version
        Mojave: `Apple LLVM version 10.0.1 (clang-1001.0.46.4)`
`Target: x86_64-apple-darwin18.7.0`
`Thread model: posix`
`InstalledDir: /Library/Developer/CommandLineTools/usr/bin`
        Catalina: `Apple clang version 11.0.0 (clang-1100.0.33.17)`
`Target: x86_64-apple-darwin19.3.0`
`Thread model: posix`
`InstalledDir: /Library/Developer/CommandLineTools/usr/bin`

**Additional context**
Add any other context about the problem here.

toucan12, 2020-04-09:
I have been experiencing the same symptoms since yesterday.
I don't know if it is the right way, but if I make the following modifications in the Makefile, it compiles normally.

In line 54 of client/Makefile,

**Before**
`CFLAGS ?= -Wall -Werror -g -O3`

**After**
`CFLAGS ?= -Wall -Werror -Wno-missing-braces -g -O3`

iceman1001, 2020-04-09:
.... or just  add a brace ...  ```{{0}}```  instead of ignoring it...

toucan12, 2020-04-09:
I've already tried it, but I get an error.

[image]

any ideas?

closed 2020-04-09 by iceman1001